OpenIntelWireless/IntelBluetoothFirmware #104: Can't load kext because of "invalid signature"

wveilleux, 2020-05-07:
**Description**
I can't load that Kext because it seem there is an invalid signature

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel 9560
- Product ID: 
- macOS Version: Catalina (10.15.4)

**Info in System Report - Bluetooth**
sudo kextutil -v /Library/Extensions/IntelBluetoothFirmware.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signature (-67062) allowed: <OSKext 0x7f8bd940f960 [0x7fff85d1b8c0]> { URL = "file:///Library/Extensions/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: not code signed
/Library/Extensions/IntelBluetoothFirmware.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Library/Extensions/IntelBluetoothFirmware.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
/Library/Extensions/IntelBluetoothFirmware.kext successfully loaded (or already loaded).

**Logs**
> Use `log show --last boot | grep IntelBlue` 

2020-05-07 13:11:43.587319-0400 0x184fd    Default     0x0                  8233   0    sudo: veilleux : TTY=ttys000 ; PWD=/Users/XXXXXX ; USER=root ; COMMAND=/sbin/kextunload /Library/Extensions/IntelBluetoothInjector.kext
2020-05-07 13:11:54.778484-0400 0x18553    Default     0x0                  8241   0    sudo: veilleux : TTY=ttys000 ; PWD=/Users/XXXXXX ; USER=root ; COMMAND=/sbin/kextload -v /Library/Extensions/IntelBluetoothInjector.kext
2020-05-07 13:11:54.826063-0400 0x2d0      Error       0x0                  63     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600000e4edc0 [0x7fff85d1b8c0]> { URL = "file:///Library/Extensions/IntelBluetoothInjector.kext/", ID = "com.zxystd.IntelBluetoothInjector" }

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

Menchen, 2020-05-07:
`/Library/Extensions/IntelBluetoothFirmware.kext successfully loaded (or already loaded).`
So what's the problem?
Invalid signature will be there because you need Apple cert to sign them. So there's no bug, because there's nothing we can do about it. You just need to disable SIP,

wveilleux, 2020-05-07:
SIP is disable, see with this command:
csrutil status
Configuration:
	Apple Internal: disabled
	Kext Signing: disabled
	Filesystem Protections: disabled
	Debugging Restrictions: disabled
	DTrace Restrictions: disabled
	NVRAM Protections: disabled
	BaseSystem Verification: disable

I force to load the kext with kextload so it's why you have that message. But from a new booted system, I don't have anything with : log show --last boot | grep IntelFirmware

I'm using Clover Bootloader and I put that Kext into the CLOVER kext partition and I have this with this command: 

8:908  0:001  Use origin smbios table type 1 guid.
8:913  0:004  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Other
8:913  0:000  ->Extra kext: EFI\CLOVER\kexts\Other\IntelBluetoothInjector.kext (v.1.0.3)
8:922  0:009  ->Extra kext: EFI\CLOVER\kexts\Other\IntelBluetoothFirmware.kext (v.1.0.3)
8:936  0:014  ->Extra kext: EFI\CLOVER\kexts\Other\RtWlanU.kext (v.1830.32.b7)
9:037  0:101  ->Extra kext: EFI\CLOVER\kexts\Other\RtWlanU1827.kext (v.1827.4.b36)
9:090  0:052  ->Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.3.8)
9:103  0:012  ->Extra kext: EFI\CLOVER\kexts\Other\VirtualSMC.kext (v.1.1.2)
9:118  0:015  ->Extra kext: EFI\CLOVER\kexts\Other\USBInjectAll.kext (v.0.7.1)
9:468  0:349  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.4.3)
9:478  0:010  ->Extra kext: EFI\CLOVER\kexts\Other\IntelMausi.kext (v.1.0.2)
9:489  0:010  ->Extra kext: EFI\CLOVER\kexts\Other\FakePCIID_Intel_HDMI_Audio.kext (v.1.3.15)
9:500  0:010  ->Extra kext: EFI\CLOVER\kexts\Other\FakePCIID.kext (v.1.3.15)
9:508  0:008  ->Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.4.8)

wveilleux, 2020-05-07:
BTW, thank you very much for that Intel Bluetooth Kext" I hope I'll be able to use it. I'm not an advanced internal MacOS, more Linux... :)

Menchen, 2020-05-07:
Check your usb map, there should be Bluetooth in settings without the Kext

wveilleux, 2020-05-08:
Sorry but what do you mean by checking my USB map ? Here is the output of slush command (from brew):
lsusb
Bus 020 Device 003: ID 1058:1140 Western Digital Technologies, Inc. My Book 1140  Serial: 574341575A31363239343134
Bus 020 Device 001: ID 05e3:0608 Genesys Logic, Inc. USB2.0 Hub 
Bus 020 Device 005: ID 05ac:1006 Apple Inc. Keyboard Hub  Serial: 000000000000
Bus 020 Device 011: ID 05ac:0250 Apple Inc. Apple Keyboard 
Bus 020 Device 007: ID 051d:0002 American Power Conversion Back-UPS ES 500 FW:801.e5.D USB FW:e5  Serial: JB0439008704
Bus 020 Device 006: ID 0c60:0009 Apogee Electronics Corp Jam 
Bus 020 Device 004: ID 05e3:0608 Genesys Logic, Inc. USB2.0 Hub 
Bus 020 Device 010: ID 058f:9520 Alcor Micro, Corp. Watchdata W1981 Smart Card Reader  Serial: 5&53D460&0&1
Bus 020 Device 009: ID 046d:c51b Logitech Inc. USB Receiver 
Bus 000 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub 
Bus 000 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. USB 10/100/1000 LAN  Serial: 0010000A7
Bus 000 Device 003: ID 059f:107c LaCie d2 TBT2 USB 3.0  Serial: 00021000890A
Bus 000 Device 001: ID 05e3:0610 Genesys Logic, Inc. USB2.1 Hub 
Bus 000 Device 006: ID 1a40:0101 TERMINUS TECHNOLOGY INC. USB 2.0 Hub 
Bus 000 Device 007: ID 2109:0102 VIA Labs, Inc. USB 2.0 BILLBOARD  Serial: 0000000000000001
Bus 000 Device 005: ID 0bda:4835 Realtek Semiconductor Corp. USB Audio  Serial: 201405280001
Bus 000 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub 
Bus 000 Device 001: ID 1d6b:CIAR Linux Foundation USB 3.1 Bus 

My Bluetooth device is a built-in Intel 9560 on an Intel NUC8

tvhung83, 2020-05-08:
@wveilleux it means you need to make sure that USB port mapping is working properly. In my case, when it works, I can see internal USB ports like this:
[image]
Check this guide: https://dortania.github.io/USB-Map-Guide/
If it didn't work, try different SMBIOS. I used to use `MacPro6,1`, and there is no way I can see internal (header) ports. But it's fixed as soon as I switch to `iMac19,1` (which is recommended in many guides).

wveilleux, 2020-05-08:
OK, thanks. But at boot, I have this:

0:100  0:000  === [ GetDevices ] ==============================
0:100  0:000  GOP found at: PciRoot(0x0)/Pci(0x2,0x0)/AcpiAdr(0x80010100)
0:100  0:000  PCI (00|00:00.00) : 8086 3ED0 class=060000
0:100  0:000  PCI (00|00:02.00) : 8086 3EA5 class=030000
0:100  0:000   - GOP: Provided by device
0:100  0:000   - GFX: Model=Intel Iris Plus Graphics 655 (Intel)
0:100  0:000  PCI (00|00:08.00) : 8086 1911 class=088000
0:100  0:000  PCI (00|00:12.00) : 8086 9DF9 class=118000
0:100  0:000  PCI (00|00:14.00) : 8086 9DED class=0C0330
0:100  0:000  PCI (00|00:14.02) : 8086 9DEF class=050000
0:100  0:000  PCI (00|00:14.03) : 8086 9DF0 class=028000
0:100  0:000   - WIFI: Vendor=32902Intel
0:100  0:000  PCI (00|00:16.00) : 8086 9DE0 class=078000
0:100  0:000  PCI (00|00:17.00) : 8086 9DD3 class=010601
0:100  0:000  PCI (00|00:1C.00) : 8086 9DB8 class=060400
0:100  0:000  PCI (00|00:1C.04) : 8086 9DBC class=060400
0:100  0:000  PCI (00|02:00.00) : 8086 15DA class=060400
0:100  0:000  PCI (00|03:00.00) : 8086 15DA class=060400
0:100  0:000  PCI (00|03:01.00) : 8086 15DA class=060400
0:100  0:000  PCI (00|03:02.00) : 8086 15DA class=060400
0:100  0:000  PCI (00|6C:00.00) : 8086 15DB class=0C0330
0:100  0:000  PCI (00|00:1D.00) : 8086 9DB0 class=060400
0:101  0:000  PCI (00|6D:00.00) : 126F 2262 class=010802
0:101  0:000  PCI (00|00:1F.00) : 8086 9D84 class=060100
0:101  0:000  PCI (00|00:1F.03) : 8086 9DC8 class=040380
0:101  0:000  PCI (00|00:1F.04) : 8086 9DA3 class=0C0500
0:101  0:000  PCI (00|00:1F.05) : 8086 9DA4 class=0C8000
0:101  0:000  PCI (00|00:1F.06) : 8086 15BE class=020000

I see 8086 as IDs instead of 8087. Can it be my problem ? BTW, I tried to change my SMBIOS to iMac19,1 and I can't both with that SMBIOS.

tvhung83, 2020-05-09:
@wveilleux please double check your USB mapping. Also, I don't know where did you get `0x8087` Vendor ID, mine is `0x8086` as well (same model with yours).

wveilleux, 2020-05-09:
You know what, I have a NUC8i5BEH and I think I have a problem with my Bluetooth device because even if I boot with Linux, Windows or MacOS, no Bluetooth device is detect (and I installed the Intel driver for Windows 10). So, thanks very, very much for your time and I really apologize. I should have think about it.... I will try to fix this first. Regards

wveilleux, 2020-06-03:
Finally, I got a brand new NUC8i5BEH from INTEL and Bluetooth is working perfectly :) The bluetooth hardware was broken...

closed 2020-06-03 by wveilleux