OpenIntelWireless/IntelBluetoothFirmware #105: Is someone actively working on this Kext? I haven't seen an update being pushed.

joostiphone, 2020-05-08:
**Description**
> A clear and concise description of what the bug is

**Environment**

- Kext Version: 
- WiFi Card Model: 
- Product ID: 
- macOS Version: 

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

zxystd, 2020-05-08:
Keep updating of course, so what's wrong with this kext?

joostiphone, 2020-05-13:
Sorry, I didn't notived your answer. Thank you for replying. 

It seems that I have some stability issues regarding unlocking my Mac (Hackintosh) with my Apple Watch. Only the first time after a reboot it works, and after that, it stops (doesn't recognize the Apple Watch anymore). Any ideas/experiences wit this?

zxystd, 2020-05-13:
Sorry I dont't have apple watch to test. Can you repeat the question? If so, Restart, then repeat the problem, and tell me what steps have been taken to unlock or operate the computer at which time point, then send the log to me, and I will check what happened according to the corresponding time point of the log.

joostiphone, 2020-05-13:
Regarding: 

`log show --last boot | grep IntelFir
`
Is the output being generated in Terminal itself? Or in a file? In Terminal I don't get a response/log.

StefanLobbenmeier, 2020-06-03:
> Regarding:
> 
> `log show --last boot | grep IntelFir `
> Is the output being generated in Terminal itself? Or in a file? In Terminal I don't get a response/log.

This should print any error to the terminal. No output means that there was no error in the output of
> `log show --last boot

Containing 
> IntelFir

williambj1, 2020-07-13:
This issue hasn't been active for one month, required logs are  not given as well. Feel free to reopen.

closed 2020-07-13 by williambj1