OpenIntelWireless/IntelBluetoothFirmware #106: Intel Bluetooth works without kext?

tvhung83, 2020-05-09:
**Description**

Once the kext is loaded, I can't unload it, ie. make Intel Bluetooth to stop appearing.

The problem Intel Bluetooth is, my bluetooth audio input is not working, hence I want to get back to my bluetooth USB adapter, but I can't find a way to make it stop working.

I removed the kext entry in OC `config.plist`, move the kext outside of `Kexts` folder. Still, System Information keeps saying my Bluetooth is Intel and old connected devices keep working as if I was using the kext.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel 9560
- Product ID: 
- macOS Version: 10.5.4

**Info in System Report - Bluetooth**
```
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	[REDACTED]
  Address:	[REDACTED]
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AAA
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  RK-Bluetooth keyboard:
  Address:	[REDACTED]
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Broadcom Bluetooth Wireless Keyboard SDP Server, Broadcom Bluetooth Wireless  Keyboard                        , Broadcom Bluetooth Wireless Keyboard PnP Server
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x2211)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x011B
  Vendor ID:	0x05AC
  Product ID:	0x024F
  Class of Device:	0x05 0x10 0x0540
  AFH:	On
  AFH Map:	FEFF370000000
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	Yes
  SBH24:
  Address:	[REDACTED]
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	Sony Mobile Remote Control Service, Headset-HS, Sony Mobile Audio Service, Sony Mobile Hands free Service
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Unknown (0x8, 0x3232)
  Bluetooth Core Spec:	4.2
  Firmware Version:	0x1000
  Vendor ID:	0x0094
  Product ID:	0x5880
  Class of Device:	0x00 0x00 0x0000
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```

**Logs**
```log
# empty, it proves that I'm not loading the kext
```

**Kernel Panic Logs**
```
none
```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

tvhung83, 2020-05-09:
FWIW, I still can make it stop working by removing USB port mapping, but that's not what I want.

Menchen, 2020-05-09:
macOs can only use the first bt device it founded... so no hot swapping.

tvhung83, 2020-05-09:
@Menchen I didn't mean hot swap, I mean how can I remove Intel bluetooth? I've removed the kext from `config.plist` as well as delete kext file, but now Intel bluetooth device keeps appearing, that's my question.

Menchen, 2020-05-09:
Disable int bt device using custom usb map.

tvhung83, 2020-05-09:
Ah, let me try it and update here.

tvhung83, 2020-05-11:
I decided to remove USB mapping, so close this for now.

closed 2020-05-11 by tvhung83