OpenIntelWireless/IntelBluetoothFirmware #107: Audio Input is not working, device got disconnected when switching input devices

tvhung83, 2020-05-09:
**Description**
First of all, thank you for your hard work and this wonderful kext!!! I don't even think that my Intel bluetooth can work one day.

This is just one final piece to make it perfect. My apologies if it's taking your time.

I used my [bluetooth headset](https://www.sony-asia.com/electronics/accessories/stereo-bluetooth-headset-sbh24) a lot for call conference. Audio output works excellent. But unfortunately, audio input is not working, even though the device showed up in Input tab of Sound pref panel.
[image]

I tried to tap the micro but Input level is simply not moving. And when I tried to switch between Input devices, the headset got disconnected after a few seconds. UI of pref panel was hanging a little bit too.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel 9560
- Product ID: `0x0AAA`
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
```
Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	[REDACTED]
  Address:	[REDACTED]
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AAA
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  RK-Bluetooth keyboard:
  Address:	[REDACTED]
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Broadcom Bluetooth Wireless Keyboard SDP Server, Broadcom Bluetooth Wireless  Keyboard                        , Broadcom Bluetooth Wireless Keyboard PnP Server
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x2211)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x011B
  Vendor ID:	0x05AC
  Product ID:	0x024F
  Class of Device:	0x05 0x10 0x0540
  AFH:	On
  AFH Map:	FEFF370000000
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
[... 1608 lines cut ...]
default	12:49:54.380140+0700	bluetoothaudiod	1 clients have now registered for Now Playing Notifications
default	12:49:54.380305+0700	mediaremoted	Adding client <MRDMediaRemoteClient 0x7fad4781adf0, bundleIdentifier = bluetoothaudiod, pid = 7322>
default	12:49:54.380377+0700	bluetoothaudiod	Device supports HFP
default	12:49:54.380456+0700	bluetoothaudiod	Set current codec (null)->SCO
default	12:49:54.380577+0700	bluetoothaudiod	Device supports HSP
default	12:49:54.380604+0700	bluetoothaudiod	Set current codec (null)->SCO
default	12:49:54.380840+0700	bluetoothaudiod	Created audio device SBH24
default	12:49:54.380923+0700	bluetoothaudiod	XPC: BTStopIO for SBH24
default	12:49:54.380942+0700	bluetoothaudiod	Stop IO - set input activeBeforeSleep = NO
default	12:49:54.380959+0700	bluetoothaudiod	Already stopped - return immediately
default	12:49:54.380973+0700	bluetoothaudiod	Update activity signaled:0
default	12:49:54.380987+0700	bluetoothaudiod	sendAudioStatus isAudioActive:0
default	12:49:54.381969+0700	bluetoothaudiod	Quitting now
default	12:49:54.382739+0700	coreaudiod	XPC Error remove all devices
default	12:49:54.382973+0700	coreaudiod	Remove all devices
default	12:49:54.410670+0700	mediaremoted	Client <MRDMediaRemoteClient 0x7fad4781adf0, bundleIdentifier = bluetoothaudiod, pid = 7322> invalidated.
default	12:49:54.410737+0700	mediaremoted	Removing client <MRDMediaRemoteClient 0x7fad4781adf0, bundleIdentifier = bluetoothaudiod, pid = 7322>
default	12:49:54.495973+0700	mediaremoted	[AVRoutingServer] Setting (null) to application picked routes for client <MRDMediaRemoteClient 0x7fad4781adf0, bundleIdentifier = bluetoothaudiod, pid = 7322>
default	12:49:59.343528+0700	coreaudiod	1 second delay - has input been reactivated? 0
default	12:49:59.343612+0700	coreaudiod	Request device configuration change: Sample rate to 48000
```
</details>

**Screenshots**
> If applicable, add screenshots to help explain your problem.

f0rdream, 2020-05-09:
Same for the ax200 model when I connect a Sony Xm3 and use the mircophone.

I will post related info later

f0rdream, 2020-05-09:
**Description**
When I use only the output, then everything works like a charm. Thanks for your great job!

But when I open Sound prefPanel and click the input tab, UI freezes. 

If further infomation is needed, please let me know, I will gladly provide them.

**Environment**
- Kext Version: 1.0.3
- WiFi Card Model: Intel ax200
- Product ID: 0x0029
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**

```
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	wangyang2’s iMac
  Address:	24-41-8C-04-9D-9E
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	Off
  Vendor ID:	0x8087
  Product ID:	0x0029
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off
  Devices (Paired, Configured, etc.):
  MacBook Pro (2):
  Address:	CD-31-FD-15-D3-CE
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	
  Paired:	No
  Configured:	Yes
  Connected:	No
  Manufacturer:	Broadcom (0x9, 0x4101)
  Bluetooth Core Spec:	5.0
  Class of Device:	0x00 0x00 0x0000
  WH-1000XM3:
  Address:	38-18-4C-2B-75-B1
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	Wireless iAP, Hands-Free unit, Amazon Alexa, AudioWear, Headset, CSR GAIA‚Ñ¢, AudioWear Voice
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Cambridge Silicon Radio (0x7, 0x328E)
  Bluetooth Core Spec:	4.1
  Firmware Version:	0x0422
  Vendor ID:	0x054C
  Product ID:	0x0CD3
  Class of Device:	0x00 0x00 0x0000
[... 80 lines cut ...]
May 10 00:14:54 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9725): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/SharedFrameworks/DTDeviceKitBase.framework/Versions/A/XPCServices/com.apple.dt.Xcode.TCPRelayService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
May 10 00:14:54 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9725): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/Frameworks/IDEKit.framework/Versions/A/XPCServices/IDETouchBarSimulatorService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
May 10 00:14:54 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9725): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/SharedFrameworks/LLDB.framework/Versions/A/XPCServices/RootDebuggingXPCService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
May 10 00:14:54 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9725): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Xcode/PrivatePlugIns/IDEOSXSupportCore.ideplugin/Contents/XPCServices/com.apple.dt.XcodeMacLocationSimulation.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex

2020-05-09 16:18:41 +0000 Handle: 0x101 disconnected, reason: 0x16
May 10 00:19:55 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.deleted.9222): Service "com.apple.SafariTechnologyPreview.CacheDeleteExtension" tried to register for endpoint "com.apple.Safari.CacheDeleteExtension" already registered by owner: com.apple.Safari.CacheDeleteExtension
May 10 00:19:56 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9774): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/SharedFrameworks/DTDeviceKitBase.framework/Versions/A/XPCServices/com.apple.dt.Xcode.TCPRelayService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
May 10 00:19:56 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9774): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/Frameworks/IDEKit.framework/Versions/A/XPCServices/IDETouchBarSimulatorService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
May 10 00:19:56 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9774): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/SharedFrameworks/LLDB.framework/Versions/A/XPCServices/RootDebuggingXPCService.xp [... cut ...]
May 10 00:25:00 Fordream-PC-realtek com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.9870): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Xcode/PrivatePlugIns/IDEOSXSupportCore.ideplugin/Contents/XPCServices/com.apple.dt.XcodeMacLocationSimulation.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
```

atar13, 2020-05-12:
Same issue for device-id 0x8087.

zxystd, 2020-05-13:
Audio bug has nothing to do with this driver. It just uploads firmware to make Bluetooth hardware work. The rest is handed over to Apple's Bluetooth driver.

tvhung83, 2020-05-13:
@zxystd the same audio device works with bluetooth dongle (Bluetooth 4.0), so it shouldn't be Apple's Bluetooth driver?
```
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	[REDACTED]
  Address:	[REDACTED]
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Cambridge Silicon Radio
  Transport:	USB
  Firmware Version:	8891
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x0A12
  Product ID:	0x0001
  Bluetooth Core Spec:	4.0 (0x6)
  HCI Revision:	0x22BB
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x22BB
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  RK-Bluetooth keyboard:
  Address:	[REDACTED]
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Broadcom Bluetooth Wireless Keyboard SDP Server, Broadcom Bluetooth Wireless  Keyboard                        , Broadcom Bluetooth Wireless Keyboard PnP Server
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x2211)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x011B
  Vendor ID:	0x05AC
  Product ID:	0x024F
  Class of Device:	0x05 0x10 0x0540
  AFH:	On
  AFH Map:	000C07F600C07F
  RSSI:	-66
  Role:	Master
  Connection Mode:	Sniff Mode
  Interval:	11.25 ms
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	Yes
  SBH24:
  Address:	[REDACTED]
  Major Type:	Audio
  Minor Type:	Headset
  Services:	Sony Mobile Remote Control Service, Headset-HS, Sony Mobile Audio Service, Sony Mobile Hands free Service
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Unknown (0x8, 0x3232)
  Bluetooth Core Spec:	4.2
  Firmware Version:	0x1000
  Vendor ID:	0x0094
  Product ID:	0x5880
  Class of Device:	0x04 0x01 0x240404
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

tvhung83, 2020-05-13:
When I compare device info between Intel vs external dongle:
- Intel:
```
  Major Type:	Miscellaneous
  Minor Type:	Unknown
```
- Dongle:
```
  Major Type:	Audio
  Minor Type:	Headset
```

zxystd, 2020-05-13:
Apple's Bluetooth driver is customized for Broadcom chip, which may include manufacturer's customization HCI commands, and these chips are different from Intel chips, so there will be more or less compatibility problems. If you know some HCI protocols, you can use Apple's `packetlogger` tool to grasp HCI packets and compare whether they are different. Unfortunately, I do not have an audio input device.

tvhung83, 2020-05-13:
Yeah, looks like a bug of Apple's driver: https://www.reddit.com/r/hackintosh/comments/7clx0s/not_able_to_use_bluetooth_headset_for_audio_input/
Well, I do not know HCI protocol, but will try to take a look.
Quick question though, if it's compatibility issue, is it possible to fix? (I only mean the feasibility, if it's zero, it's totally fine to close this).

zxystd, 2020-05-13:
Of course yes, by packetlogger and reverse driver we can know what is happening on with audio input, and we can fix it just like a patch(or lilu). I am now working with intel wifi working, so not pay too much time on this.

ZambraDesigner, 2020-05-15:
> Yeah, looks like a bug of Apple's driver: https://www.reddit.com/r/hackintosh/comments/7clx0s/not_able_to_use_bluetooth_headset_for_audio_input/
> Well, I do not know HCI protocol, but will try to [...]

https://github.com/zxystd/IntelBluetoothFirmware/issues/108

tvhung83, 2020-05-16:
@ZambraDesigner thanks for sharing, I tried both `BrcmPatchRAM` & `BrcmPatchRAM3` but it makes no difference at all (with or without `IntelBluetoothFirmware.kext`).

huzzam, 2020-05-29:
same issue here, headset is soundbuds slim, intel card id is 0x0aaa

f0rdream, 2020-05-29:
After upgraded to 10.15.5, audio input works!

tvhung83, 2020-05-29:
> After upgraded to 10.15.5, audio input works!

aww, congrats! I guess I just have a bad luck 😭

f0rdream, 2020-06-01:
> After upgraded to 10.15.5, audio input works!

By the way, I've also upgraded OpenCore to 0.58, maybe this is the reason that I can use the audio input. Hope this could help.

tvhung83, 2020-06-01:
I'm on OC 0.5.8 as well, have no time to dig more yet, but probably due to my model.

tvhung83, 2020-06-02:
> > After upgraded to 10.15.5, audio input works!
> 
> By the way, I've also upgraded OpenCore to 0.58, maybe this is the reason that I can use the audio input. Hope this could help.

@f0rdream could you share your `config.plist`? Just the `Kernel` -> `Add` section (load kexts) will do. Today I installed latest OS update, upgrade to OC 0.5.9, and tried a numerous combinations of kexts without luck (`IntelBluetoothInjector`, `BcrmPatchRAM*`, etc.).

tvhung83, 2020-07-07:
I realized that my Xiaomi bluetooth speaker (which has audio input) works _most_ of the time (sometimes it stops working, I still can't consistently reproduce). So the issue is not at Intel bluetooth device. Closing this.

huzzam, 2020-07-07:
I'm still having this issue, on v1.1.1, OC 0.5.8, Mojave 10.14.6

tvhung83, 2020-07-07:
> I'm still having this issue, on v1.1.1, OC 0.5.8, Mojave 10.14.6. Recommend not closing the issue.

It looks like a bug of macOS to me, you can try different devices and see it for yourself. The fact that audio input works for some and worked with my bluetooth speaker proves that the issue is not at Intel firmware. I plan to buy a different headset soon. Anyways, I will reopen and let @zxystd decides then.

williambj1, 2020-07-07:
Hmmm, I don't think this issue is related to this kext since it's just a firmware uploader. Please report this issue exists or not when you directly reboot from Windows/Linux without this kext installed.

huzzam, 2020-07-07:
> Hmmm, I don't think this issue is related to this kext since it's just a firmware uploader. Please check whether this issue exists or not when you directly reboot from Windows/Linux without this kext installed.

i don't have any issues with the same headphones with the same bluetooth adapter in windows or linux.

williambj1, 2020-07-07:
@huzzam I mean rebooting from Windows/Linux but not in Windows/Linux.

The core part of macOS bluetooth is still handled by Apple’s Bcrm kexts, Intel cards work with Apple’s kexts because they all use the standard HCI command protocol, but it’s likely to cause compatibility issues as well.

Once again, IntelBluetoothFirmware.kext is just a pure firmware uploader, it has nothing to do with your headphones.

tvhung83, 2020-07-08:
> Apple’s Bcrm kexts
yeah, I also think this is the cause. Since it works with some other devices.

zxystd, 2020-07-08:
> > Apple’s Bcrm kexts
> 
> yeah, I also think this is the cause. Since it works with some other devices.
> 
> > Please check whether this issue exists or not when you directly reboot from Windows/Linux without this kext installed.
> 
> You mean we can remove this kext if we directly reboot from Windows/Linux? Does that mean the firmware will remain after a warm reboot from Windows/Linux?

Yes.

tvhung83, 2020-07-08:
> Yes.

Below is my reproduce steps:
1. Disable the kext from OpenCore
2. Set bootloader to Windows on next boot
3. Shut down completely (I even pulled the power plug)
4. Turn on and boot to Windows: confirm that audio input is working
5. Warm reboot to macOS (kext was not loaded): issue remains

So the issue is not this kext. One thing I noticed that the headset shows 2 devices when on Windows: **Headphones** (only audio output and no input, this works with stereo, meaning audio quality is very good).
[image]

The other called **Headset** (mono audio input and with audio input).

[image]

Under macOS, it only shows one device, so good chance, is builtin Bcrm kext doesn't support this type of device.

tvhung83, 2020-07-08:
@huzzam I'm going to close this as I plan to find another headset. As @zxystd has mentioned before, in order to support a device like this, we will need manual patch, and it takes time.

closed 2020-07-08 by tvhung83

williambj1, 2020-07-08:
@tvhung83 could you upload your ioreg when you're using USB Bluetooth with your headset?

tvhung83, 2020-07-08:
> @tvhung83 could you upload your ioreg when you're using USB Bluetooth with your headset?

Give me some time, I'm in the middle of something and can't restart yet. Will upload asap.

tvhung83, 2020-07-10:
@williambj1 did you find anything from ioreg?

williambj1, 2020-07-10:
Your USB Bluetooth used a CSR chip that is natively supported by Apple, unfortunately intel nowadays use their own chips.