OpenIntelWireless/IntelBluetoothFirmware #108: Mi True Wireless earphones2 don't connect

ZambraDesigner, 2020-05-09:
I am using the 2 kext contained in the EFI of https://github.com/daliansky/XiaoMi-Pro-Hackintosh and I cannot connect my Xiaomi Mi True Wireless earphones2. They are found as a bluetooth device but do not connect.

Only once did they connect but then neither audio nor microphone worked

I also report that I am unable to connect the Logitec MX3 mouse

I have a Xiaomi Mi Notebook Pro i7 16GB 2018

ZambraDesigner, 2020-05-15:
Today I tested these kext https://bitbucket.org/RehabMan/os-x-brcmpatchram/downloads/ (BrcmPatchRAM) and now the headphone microphone works but the audio in the headphones is not good. Very metallic.

If I put the ketx above in the "Other" folder where the "IntelBluetoothFirmware" ones are already present then they no longer work.

PS: with the kext "IntelBluetoothFirmware" the audio in the headphones is very good

williambj1, 2020-07-13:
Could you please confirm that you are using Bluetooth on an Intel wireless card? I'm quite confused based on your given information.

closed 2020-07-17 by williambj1