OpenIntelWireless/IntelBluetoothFirmware #109: Please use english for the log messages

Ecco, 2020-05-11:
**Description**

Currently a bunch of log messages are in Chinese, making it impossible to understand for a bunch of users 😄

** Example **

```
kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
```

Menchen, 2020-05-11:
I can assure you that theses logs won’t help your problem as it just general Kext logging. But I agreed with you, it may be harder to debug.

Ecco, 2020-05-11:
Now that I think about it, it should also apply to GitHub issues too…

Menchen, 2020-05-11:
Well I think that’s a little harder problem as not everyone knows English and google translating it would be harder to understand.

aluveitie, 2020-05-11:
With Google Translate available (and even integrated into Chrome) it shouln't be that big of an issue if some things are in Chinese. If in doubt just ask :)

Ecco, 2020-05-11:
Sure. I still believe this isn't a good decision to code/log in any non-english languages. Just makes interacting with the project so much harder for a bunch of people.

zxystd, 2020-05-13:
OK, a good idea.

mojo2012, 2020-07-12:
> Well I think that’s a little harder problem as not everyone knows English and google translating it would be harder to understand.

I think there are more people understanding english than chinese! like 6 billion vs 1 billion?

Menchen, 2020-07-12:
@mojo2012 just looking into Wikipedia, Mandarin Chinese is the language with the most native speakers. Source: https://en.m.wikipedia.org/wiki/List_of_languages_by_number_of_native_speakers

Ecco, 2020-07-12:
Sure, but https://en.wikipedia.org/wiki/List_of_languages_by_total_number_of_speakers 😄

Also, most engineering tools are in english, and so is GitHub itself for instance. My point is that picking a language will obviously put some people at a disadvantage, but english is the _de facto_ standard when it comes to software engineering. In other words, most (all?) people reading log messages will have some understanding of english, which isn't the case for mandarin. So picking english would help make this project more universal.

zxystd, 2020-07-12:
I have change the logs into english in version 1.1.1, thanks for your advice.

closed 2020-07-12 by zxystd