OpenIntelWireless/IntelBluetoothFirmware #111: IntelBluetoothFirmware doesn't work

EETagent, 2020-05-16:
**Description**
> A clear and concise description of what the bug is

 IntelBluetoothFirmware ver 1.0.3 doesn't prints any log. I am unable to connect to any Bluetooth device. I am able to see them in Bluetooth settings. 

IntelBluetoothFirmware 1.0.2 prints same 2 lines every time. It's possible to connect to Magic Mouse, but connection fails at some point and then doesn't work. Longest session was probably 20 sec. 

During OpenCore boot
```
IntelFirmware: interrupt wait
IntelFirmware: "strange symbols"
IntelFirmware: can not proceed response
IntelFirmware: interrupt continue
IntelFirmware: HCI_OP_INTEL_EVENT_MASK
IntelFirmware: interrupt wait
IntelFirmware: interrupt continue
IntelFirmware: End download
IntelFirmware: "strange symbols" (-536878165) operation was aborted)
IntelFirmware: Driver Stop()
```
There is probably something strange going on with my EFI, I don't know what. USB ports are mapped, everything is pretty much golden.

**Environment**

- Kext Version: 1.0.2 and 1.0.3
- WiFi Card Model: AC 8265
- Product ID: 0x0a2b
- macOS Version: 10.15.3

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version:	7.0.3f5
  Hardware, Features, and Settings:
  Name:	MacBook Pro
  Address:	D4-3B-04-BD-A6-36
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.256 (8.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	No
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

log 1.0.2
```
2020-05-16 09:26:58.576421+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:26:58.576632+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:41.399197+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:41.399818+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:41.900144+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:41.900333+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:42.400901+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:42.401493+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:42.909722+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:42.909957+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:43.410538+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:43.410760+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:43.912819+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:43.913036+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
2020-05-16 09:29:44.413289+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=0
2020-05-16 09:29:44.413536+0200 0x135      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: setPowerState powerStateOrdinal=1
etc.
```
log 1.0.3
```
Nothing
```

**Screenshots**
> [image]

> [image]

williambj1, 2020-05-16:
AFAIK, 1.0.3 replaced the builtin firmware with a newer version. Try the older versions from Linux Firmware, recompile the kext, and see if that helps.

EETagent, 2020-05-16:
@williambj1  I tried direct rebooting from the Windows and Bluetooth started working, probably because Windows injected firmware to the Bluetooth card first. Then I performed full shutdown and booted to the macOS directly, Magic Mouse and Bluetooth still works.

 I don't think it's all OK though, there are still no grep IntelFir logs and there is still Driver Stop() line in the OpenCore boot log ( I don't know if its normal ). Could firmware persist even after shutdown? Would it be worth compiling kext with newer firmware files? Latest release is already more than month old so Intel probably released some update. Thanks

EETagent, 2020-05-17:
Today I started my computer. For five minutes after boot, Bluetooth was shown as unreachable, then it started to work. Even the logs appeared in the terminal. It's weird, because I didn't changed anything from the time when I posted this issue. Things always start to work when I decide to ask for help. 

Before I close this, are these logs OK? Thanks

'''
2020-05-17 10:00:12.416104+0200 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-05-17 10:00:12.418735+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-05-17 10:00:12.418744+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-05-17 10:00:12.418753+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
2020-05-17 10:00:12.418823+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-05-17 10:00:13.921135+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-05-17 10:00:13.922072+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-17 10:00:13.922082+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-17 10:00:13.922089+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-17 10:00:13.922097+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-05-17 10:00:13.922157+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-05-17 10:00:13.922550+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-05-17 10:00:13.922917+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-05-17 10:00:13.923199+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-05-17 10:00:13.923208+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-05-17 10:00:13.924301+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-17 10:00:13.924890+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.0 build 26 week 38 2015
2020-05-17 10:00:13.924898+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: supect device firmware: ibt-12-18-0.sfi
2020-05-17 10:00:13.924914+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-17 10:00:13.924919+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-05-17 10:00:13.925152+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-17 10:00:13.925891+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 16
2020-05-17 10:00:13.925896+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-05-17 10:00:13.925900+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-05-17 10:00:13.925903+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-05-17 10:00:13.925907+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-05-17 10:00:13.925911+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-05-17 10:00:13.926198+0200 0x15c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-12-16.sfi
2020-05-17 10:00:13.926211+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-17 10:00:13.926214+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-05-17 10:00:13.926662+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-05-17 10:00:13.927283+0200 0x442b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
[... cut ...]
2020-05-17 11:42:32.136559+0200 0x17632    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-17 11:42:32.136568+0200 0x17632    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-05-17 11:42:32.136654+0200 0x17600    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-05-17 11:42:32.136715+0200 0x17632    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
'''

ISPP2056, 2020-05-18:
感谢UP主，冷启动时好时坏，不知为什么。即使这样也满意了

closed 2020-06-18 by EETagent