OpenIntelWireless/IntelBluetoothFirmware #112: Intel Nuc5i5RYK

BasselGustav, 2020-05-19:
Hello Zxystd, 

i used your kexts in Clover all works fine but sometimes it just freezes and i must connect another mouse to turn the bluetooth off and then on again to connect magic mouse 2 and magic keyboard 2 

the onboard wifi+bluetooth card is Intel AC-7265 and here is the log i get:

Last login: Tue May 19 12:51:30 on console
Mac-mini:~ Mr_B$ log show --last boot | grep IntelFirmware
2020-05-19 12:51:16.793127+0200 0x15d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-05-19 12:51:16.793140+0200 0x15d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-05-19 12:51:16.793153+0200 0x15d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-05-19 12:51:16.793218+0200 0x15d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
Mac-mini:~ Mr_B$

BasselGustav, 2020-05-19:
one quick update:
i updated the USBInjectAll.Kext and after another reboot i got the following log which is longer than the one i got the first time:

Mac-mini:~ Mr_B$ log show --last boot | grep IntelFirmware
2020-05-19 13:29:43.796887+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-05-19 13:29:43.796900+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-05-19 13:29:43.796912+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-05-19 13:29:43.796970+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-05-19 13:29:45.298882+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-05-19 13:29:45.299654+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-05-19 13:29:45.299664+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-05-19 13:29:45.299670+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-05-19 13:29:45.299676+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-05-19 13:29:45.299728+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-05-19 13:29:45.299930+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-05-19 13:29:45.300134+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-05-19 13:29:45.300306+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-05-19 13:29:45.300310+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_RESET
2020-05-19 13:29:45.300738+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-05-19 13:29:45.314580+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-05-19 13:29:45.314598+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-05-19 13:29:45.314848+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-05-19 13:29:45.315552+0200 0xe9       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Intel device is already patched. patch num: 32
2020-05-19 13:29:45.315598+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-05-19 13:29:45.315614+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-05-19 13:29:45.315891+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-05-19 13:29:45.316544+0200 0xe9       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-05-19 13:29:45.316587+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-05-19 13:29:45.316611+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-05-19 13:29:45.316659+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

williambj1, 2020-07-13:
Is this issue still reproduceable with the latest version v1.1.1?

closed 2020-07-17 by williambj1