OpenIntelWireless/IntelBluetoothFirmware #113: Can't get the kext to load with OpenCore

MrCreeper1008, 2020-05-20:
**Description**
I injected the kext to opencore by putting it inside OC/Kext and adding it to config.plist. Nothing shows up when I enter the command to output a log. 

**Environment**

- Kext Version: latest
- WiFi Card Model: 
- Product ID: 0x8087, 0xaaaa
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

Nothing

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

No

williambj1, 2020-05-21:
I have the same card as yours and I’m using OpenCore, everything works as expected. I don’t think this is a bug, it’s certainly your configuration mistake based on your given information. Consider filling all the blanks in the template form if you want to argue with me. Thanks.

MrCreeper1008, 2020-05-21:
I can't reach my pc right now; I am commenting with the mobile app. I'll certainly full the rest of the information when i can use my pc. Thanks for the suggestion though.

MrCreeper1008, 2020-05-21:
@williambj1 info updated

MrCreeper1008, 2020-05-21:
@williambj1 when i search through my OpenCore log i found these two lines:
```
11:266 00:034 OC: Prelink injection IntelBluetoothFirmware.kext () - Success
11:269 00:002 OC: Prelink injection IntelBluetoothInjector.kext () - Success
```
It seems like they are injected succesfully.

williambj1, 2020-05-21:
Is your bluetooth working as expected and the only issue is you don't get the logs right?

Logging is a common bug in macOS (sigh), but there's an alternative way that guarantees the logs will be saved.

add `msgbuf=1048576` into boot-args
in terminal use ```sudo dmesg | grep -i IntelFir > ~/Desktop/log.txt```

> The method is provided by lvs1974, original link: https://github.com/VoodooI2C/VoodooI2C/issues/269#issuecomment-608298277

MrCreeper1008, 2020-05-21:
My Bluetooth is not working properly. No devices are visible on the list until a device requests to pair first. I paired my trackpad with the computer with a USB cable, but sometimes the trackpad will refuse to connect to the computer.

williambj1, 2020-05-21:
> No devices are visible on the list until a device requests to pair first.

AFAIK, this is expected, that's how bluetooth works. Does the device list appear the same on your phone or other devices/systems?

MrCreeper1008, 2020-05-21:
When both devices are visible, I should be able to pair from both devices. The Bluetooth list on my computer is completely empty, but the list on my phone shows my computer just fine.

williambj1, 2020-05-21:
That’s quite odd, did you try with peripherals that have different bluetooth versions, e.g. bluetooth 4.0 & bluetooth 5.0?

Does directed rebooting from Windows without powering off (aka warm reboot) **without** the kext installed have the same issue? If yes then your issue might have nothing to do with this kext since it’s just a firmware uploader, all the core parts are handled by MacOS.

MrCreeper1008, 2020-05-21:
Yep, did that and had the same issue. However, Bluetooth works just fine on Windows, and I have not seen any user having Bluetooth issues with macOS `10.15.4`.

MrCreeper1008, 2020-05-21:
Also tried to use the method of guaranteeing a log; didn't work. log.txt is empty.

williambj1, 2020-05-22:
I'm afraid that's all I could help since the device is not on my hands and I have no idea what's wrong with your configuration, good luck.

MrCreeper1008, 2020-06-03:
@williambj1 The kext was actually loaded:
```
$ kextstat | grep -v com.apple | grep IntelBluetooth
94    0 0xffffff7f86369000 0xea8000   0xea8000   com.zxystd.IntelBluetoothFirmware (1.0.3) BBD2A3B5-1B2F-3DAF-919A-D7EECD606FF1 <27 6 5 3>
```

zxystd, 2020-06-04:
Some devices such as BT 4.0 mouse can't connect because of the compatibility of Apple driver and Intel BT, not due to this Firmware Driver, you can try another device, Thanks.

closed 2020-06-04 by zxystd

MrCreeper1008, 2020-06-04:
Cool, good to know. Thanks!

MrCreeper1008, 2020-07-09:
I figured out what happened. I need to attach an antenna to my motherboard to increase signal strength.

sojab0on, 2021-03-27:
I have same issue bummer that bt 4.0 devices don't work I don't have mice with tiger then 4.0 BT tech in them 

nice thing is that my MX Master has dual tech so that will be used with a usb receiver uniting :) hope that BT 4.0 will get working soon 

Other devices like Jabra t110BT and Kove 101 headphones work fine on BT 
even tough the Tune 110BT is also BT 4.0 and the 101 is a BT4.1

johnking0099, 2022-07-03:
> I figured out what happened. I need to attach an antenna to my motherboard to increase signal strength.
👍