OpenIntelWireless/IntelBluetoothFirmware #115: Firmware Issues

Apfel, 2020-05-23:
**Description**
The Bluetooth card refuses to work until the Mac gets rebooted. This mainly occurs if the Mac booted cold, but might happen after normal reboots as well.

**Environment**
- Kext Version: 1.03
- WiFi Card Model: Intel 8260NGW (Inside ASUS PCE-AC55BT)
- Product ID: 0x0A2B (Vendor ID: 0x8087)
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
```log
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features and Settings:
  Name:	Marko’s iMac Pro
  Address:	< redacted >
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	7.256 (7.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.1 (0x7)
  HCI Revision:	0x100
  LMP Version:	4.1 (0x7)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off
  Devices (Paired, Configured, etc.):
    < I have severe doubt that it would've been necessary to show that I had just my two Windows Mixed Reality and an Xbox One controller connected >
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```
*Apologies for the atrocious tabs.*

**Logs**
Before rebooting:
```log
2020-05-23 18:03:25.484797+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-05-23 18:03:25.485073+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-05-23 18:03:25.485075+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-23 18:03:25.485077+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-05-23 18:03:25.488060+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-05-23 18:03:26.488356+0200 0x1eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-23 18:03:26.489076+0200 0x1af      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
[... 29 lines cut ...]
2020-05-23 18:18:53.089060+0200 0x1af      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-11-5.sfi
2020-05-23 18:18:53.089072+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-23 18:18:53.089076+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-05-23 18:18:53.089785+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-05-23 18:18:53.090852+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-05-23 18:18:53.094957+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-05-23 18:18:55.482966+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-05-23 18:18:55.483376+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-05-23 18:18:55.483379+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-23 18:18:55.483380+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-05-23 18:18:55.485807+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-05-23 18:18:56.486420+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-23 18:18:56.486825+0200 0x1af      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
2020-05-23 18:18:56.486829+0200 0x1af      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-05-23 18:18:56.486831+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-23 18:18:56.486833+0200 0x1e8      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
```
I sadly don't speak Chinese, and Google Translate wasn't too helpful, either.
Also, after the second reboot, the Bluetooth card stopped working again, so I chose to reboot a third time.
Just like the first time, the log command did not output anything.

A possible fix might either be to ignore this "error" altogether or to wait for the card to be done (assuming that'd be neccessary) - and, considering it crashed when rebooting sometimes, maybe additional checks to prevent the firmware from being removed would be a good idea?
A tool to reload the driver/firmware/whatever on-the-fly would be nice.

There was no kernel panic, and screenshots were not necessary.

challenger800, 2020-06-14:
Environment

Kext Version: 1.0.3
WiFi Card Model: Intel 8260NGW (Inside Thinkpad)
Product ID: 0x0A2B (Vendor ID: 0x8087)
macOS Version: 10.15.6 

When first boot, the bluetooth works normal , the log shows like  as blew
Last login: Sun Jun 14 21:56:12 on console
cheng@MacBook-Pro ~ % log show --last boot | grep IntelFirmware
2020-06-14 21:55:58.209547+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-06-14 21:55:58.209557+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-06-14 21:55:58.209567+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
2020-06-14 21:55:58.209625+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-06-14 21:55:59.710938+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-06-14 21:55:59.711475+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711478+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711480+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711482+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-06-14 21:55:59.711495+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-06-14 21:55:59.711757+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-06-14 21:55:59.712022+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-06-14 21:55:59.712256+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-06-14 21:55:59.712260+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-14 21:55:59.712853+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-14 21:55:59.713318+0800 0x13c      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Firmware revision 0.0 build 10 week 41 2018
2020-06-14 21:55:59.713344+0800 0x13c      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: firmware had been download.
2020-06-14 21:55:59.713351+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-14 21:55:59.713355+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-06-14 21:55:59.713382+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
--------------Works Normal log End

After sleep then wake up from sleep mode, the bluetooth doesn't work any more.

cheng@MacBook-Pro ~ % log show --last boot | grep IntelFirmware
2020-06-14 21:55:58.209547+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-06-14 21:55:58.209557+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-06-14 21:55:58.209567+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
2020-06-14 21:55:58.209625+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-06-14 21:55:59.710938+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-06-14 21:55:59.711475+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711478+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711480+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-14 21:55:59.711482+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-06-14 21:55:59.711495+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-06-14 21:55:59.711757+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-06-14 21:55:59.712022+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-06-14 21:55:59.712256+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-06-14 21:55:59.712260+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-14 21:55:59.712853+0800 0x480      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
[... cut ...]
2020-06-14 22:01:40.830353+0800 0x13c      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 设备固件上传完成
2020-06-14 22:01:40.830367+0800 0x13c      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-06-14 22:01:40.830377+0800 0x1bb1     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-14 22:01:40.830380+0800 0x1bb1     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
------Does't work log end

suzuke, 2020-06-25:
Please try this version in this [comment](https://github.com/zxystd/IntelBluetoothFirmware/issues/127#issuecomment-648935657).
It should fix this bug.

challenger800, 2020-06-26:
> Please try this version in this [comment](https://github.com/zxystd/IntelBluetoothFirmware/issues/127#issuecomment-648935657).
> 
> It should fix this bug.

thank you I try your version

Apfel, 2020-06-30:
I just tested your version and nothing has changed - you still have to reboot after cold-booting, and every second reboot "breaks" the driver. I didn't test anything regarding hibernation (considering it seemingly doesn't work on Ryzen-based Hackintoshes), but rebooting issues still persist.

suzuke, 2020-06-30:
> I just tested your version and nothing has changed - I still have to reboot after cold-booting, and every second reboot "breaks" the driver. I didn't test anything regarding hibernation (considering it seemingly doesn't work on Ryzen-based Hackintoshes), but rebooting issues still persist.

I think that you are in a different situation, the kext that I mentioned only fixed the bug that **challenger800** said.
I will figure out what bug you facing and trying to fix it.

zxystd, 2020-07-07:
Can you try the lastest release one(v1.1.1)?

Apfel, 2020-07-07:
I just tested version 1.1.1 and it works flawlessly. My phone connected without issues and bluetooth worked without having to reboot.

closed 2020-07-07 by williambj1