OpenIntelWireless/IntelBluetoothFirmware #116: ax200间歇性失灵

812610357, 2020-06-02:
**描述**
> 简要描述 Kext 存在的问题
Thinkpad的intel ax200，今天早上还行，用起来很稳定，下午重启了一次以后就用不了了
**环境**

- Kext 版本: 10.13
- WiFi Card 型号: ax200
- 产品 ID: 
- macOS 版本: 10.15.5

**系统报告 - 蓝牙**
> 把所有内容都粘贴到下方的两个 ``` 之间

```
未查询到信息。
```

**日志**
> 终端运行 `log show --last boot | grep IntelFir`，将得到的内容粘贴至 ```log 和 ``` 之间

```log
2020-06-02 03:22:19.372481-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-02 03:22:19.372634-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-02 03:22:19.372833-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-06-02 03:22:19.390918-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-06-02 03:22:19.391167-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-06-02 03:22:19.391635-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-06-02 03:22:19.392063-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-06-02 03:22:19.392503-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-06-02 03:22:19.392713-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-02 03:22:19.393305-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-02 03:22:19.393843-0700 0x198      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Unsupported Intel hardware platform (224)
2020-06-02 03:22:19.394029-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-02 03:22:19.413558-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-06-02 03:22:19.413856-0700 0x26f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
```

**内核崩溃日志**
> 如果出现了内核崩溃, 将 `keepsyms=1` 加入到引导参数中，然后粘贴崩溃日志到下方的两个 ``` 之间

```

```

**截图**
> 如果有必要，请在下方附加截图

812610357, 2020-06-02:
更新一下系统信息，因为现在又可以用了
```
Apple蓝牙软件版本：	7.0.5f6
  硬件、功能和设置：
  名称：	MacBook Pro
  地址：	60-F2-62-A4-35-3E
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	10.256 (10.256)
  蓝牙电源：	打开
  可被发现：	关闭
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0029
  蓝牙核心规范：	Unknown (0xA)
  HCI修正版：	0x100
  LMP版本：	Unknown (0xA)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Portable
  复合类设备：	0x38010C
  设备类（主要）：	0x01
  设备类（次要）：	0x03
  服务类：	0x1C0
  自动寻找键盘：	打开
  设备（已配对、已配置等）：
  Microsoft Sculpt Comfort Mouse：
  地址：	F0-1D-BC-EE-5A-AC
  主要类型：	Miscellaneous
  次要类型：	Unknown
  服务：	Microsoft Bluetooth Mouse SDP Server        , Microsoft Bluetooth Mouse        , Microsoft Bluetooth Mouse PnP Server        
  已配对：	是
  已配置：	是
  已连接：	是
  生产企业：	Broadcom (0x5, 0x2211)
  蓝牙核心规范：	3.0
  固件版本：	0x0129
  供应商ID：	0x045E
  产品ID：	0x07A2
  设备类：	0x00 0x00 0x0000
  AFH：	打开
  AFH映射：	FEFFF00FEFF3FFF3F
  RSSI：	0
  角色：	主设备
  连接模式：	活跃模式
  时间间隔：	0 ms
  主机可连接：	否
  支持EDR：	否
  支持eSCO：	否
  支持SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM通道：	3
  需要认证：	否
```

zxystd, 2020-06-04:
非正常关机的话，有的时候会无法使用，检查一下是不是哪里造成非正常关机了。

812610357, 2020-06-04:
检查过了，的确是在非正常关机的情况下才会出现这样的情况，昨天一整天还有今天正常开机的时候都没有出现过问题。

812610357, 2020-06-05:
正常冷启动的情况下，kext每次都会加载失败，跑码画面会显示会反复加载，随后跳过，进入系统无法调取log日志。重启之后恢复正常。

toulanren, 2020-06-06:
你好，我是神舟精盾x57a1，网卡为 ax201，我这边也出现了类似问题，开机的次数和蓝牙能否驱动有关。

第一次：能用
第二次：不能用
第三次：能用
第四次：不能用
...
以此类推。。

无论是冷启动还是重启，还是两个方式相互之前切换去启动，都是如上的情况。

jianghaizhi, 2020-06-06:
我的是启动后识别不出硬件，必须睡眠一下才可以认出来 ax200

foreveryyl, 2020-07-07:
2020-07-08 04:48:11.105313+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-07-08 04:48:11.105472+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-07-08 04:48:11.105476+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-08 04:48:11.105479+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-07-08 04:48:11.105648+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-07-08 04:48:11.106681+0800 0x162      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
2020-07-08 04:48:11.106685+0800 0x162      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-07-08 04:48:11.106703+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-08 04:48:11.120729+0800 0x162      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Device reboot done
2020-07-08 04:48:11.120734+0800 0x162      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-07-08 04:48:11.120745+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-08 04:48:11.120749+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-07-08 04:48:11.121010+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-08 04:48:11.121744+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-08 04:48:11.121752+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-07-08 04:48:11.121754+0800 0x197      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

zxystd, 2020-07-07:
> 2020-07-08 04:48:11.105313+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
> 2020-07-08 04:48:11.105472+0800 0x197 Default 0x0 0 0 kernel: [...]

the log seems works fine, so what happened?

foreveryyl, 2020-07-07:
> > 2020-07-08 04:48:11.105313+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
> > 2020-07-08 04:48:11.105472+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
> > 2020-07-08 04:48:11.105476+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
> > 2020-07-08 04:48:11.105479+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
> > 2020-07-08 04:48:11.105648+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
> > 2020-07-08 04:48:11.106681+0800 0x162 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
> > 2020-07-08 04:48:11.106685+0800 0x162 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
> > 2020-07-08 04:48:11.106703+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
> > 2020-07-08 04:48:11.120729+0800 0x162 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Device reboot done
> > 2020-07-08 04:48:11.120734+0800 0x162 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
> > 2020-07-08 04:48:11.120745+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
> > 2020-07-08 04:48:11.120749+0800 0x197  0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
> > 2020-07-08 04:48:11.121010+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
> > 2020-07-08 04:48:11.121744+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
> > 2020-07-08 04:48:11.121752+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: End download
> > 2020-07-08 04:48:11.121754+0800 0x197 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
> 
> the log seems works fine, so what happened?

蓝牙鼠标不能用，连接上了，鼠标指针没有反应，AC9560

foreveryyl, 2020-07-07:
>
蓝牙鼠标不能用，连接上了，鼠标指针没有反应，AC9560
  BT3.0 Mouse：
  地址：	12-34-E0-00-AF-43
  主要类型：	Peripheral
  次要类型：	Mouse
  服务：	Keyboard
  已配对：	是
  已配置：	是
  已连接：	是
  蓝牙核心规范：	3.0
  固件版本：	0x0001
  供应商ID：	0xAE24
  产品ID：	0x8618
  设备类：	0x05 0x20 0x0580
  AFH：	打开
  AFH映射：	E620FE00E0FF1FFE3F
  RSSI：	0
  角色：	主设备
  连接模式：	活跃模式
  时间间隔：	0 ms
  主机可连接：	否
  支持EDR：	否
  支持eSCO：	否
  支持SSP：	否

foreveryyl, 2020-07-07:
主要类型：	Miscellaneous
  次要类型：	Unknown
  设备类：	0x00 0x00 0x0000

foreveryyl, 2020-07-07:
主要类型：	Miscellaneous
  次要类型：	Unknown
  设备类：	0x00 0x00 0x0000

如果是冷启动就是上面三行的内容

zxystd, 2020-07-12:
最新版的1.1.1试过了吗？

closed 2020-10-30 by zxystd