OpenIntelWireless/IntelBluetoothFirmware #118: KEXT not able to load

Parthvsquare, 2020-06-04:
**Description**
After clover update the kext was not able to load

**Environment**

- Kext Version: latest 1.0.3
- WiFi Card Model: 8265d2w
- Product ID: 
- macOS Version: 10.15.5 (19F101)

**Info in System Report - Bluetooth**
nothing 

**Logs**
(base) parthvsquare@MacBook-Pro ~ % log show --last boot | grep IntelFir
2020-06-04 16:43:30.994738+0530 0x26a      Error       0x0                  101    0    kextd: (IOKit) [com.apple.kext:kextlog] Request from non-root process 'kextload' (euid 501) to load /Users/parthvsquare/IntelFirmware - not in extensions dirs or filesystems folder.
(base) parthvsquare@MacBook-Pro ~ % 

**Kernel Panic Logs**
nada

**Screenshots**
no bluetooth seen
[image]

weidongbcn, 2020-06-07:
chmod

zxystd, 2020-06-08:
Check your USB.

James-Zhu-CA, 2020-06-08:
I have the same issue, how to deal with USB, can you share some links? @zxystd

Parthvsquare, 2020-06-08:
> I have the same issue, how to deal with USB, can you share some links? @zxystd

use hackintool.app 
it will help you remap your usb

closed 2020-06-08 by Parthvsquare