OpenIntelWireless/IntelBluetoothFirmware #119: Bluetooth devices discovered but cannot connect to them.

joepool, 2020-06-04:
Bluetooth Shows up in system preferences and system information and it can recognise devices, however, why trying to connect to a device it does not work.

**Environment**

- Kext Version: 1.03
- WiFi Card Model: 0x8087
- Product ID: 0x0025
- macOS Version: 10.13.6

**Info in System Report - Bluetooth**

```
  Apple Bluetooth Software Version:	6.0.7f16
  Hardware, Features, and Settings:
  Name:	Joe’s Mac Pro
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	On
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0025
  HCI Version:	 (0xA)
  HCI Revision:	0x100
  LMP Version:	 (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
```

**Logs**
```log
Joes-Mac-Pro:~ joe$ log show --last boot | grep IntelFirmware
2020-06-04 21:29:19.379268+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-04 21:29:19.380102+0100 0x237      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 设备固件上传完成
2020-06-04 21:29:19.380106+0100 0x237      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-06-04 21:29:19.380109+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-04 21:29:19.380112+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-06-04 21:29:19.380605+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-04 21:29:19.391103+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-04 21:29:19.391110+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-06-04 21:29:19.392261+0100 0x237      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-06-04 21:29:19.392307+0100 0x263      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

```

weidongbcn, 2020-06-13:
你能描述再清楚点吗？连接它什么不起作用？
如果可以配对，发送就应该马上有用的。接收需要打开蓝牙共享，在面板的共享里头设置。

LiuLiujie, 2020-07-17:
相同的问题，ac9560可以搜索到蓝牙设备，但无法连接，蓝牙设备下方小字显示“已连接”，蓝牙icon也有三点标志。但蓝牙设备右侧的菊花仍然在转且蓝牙设备无法使用，一段时间后又自动断开了

williambj1, 2020-07-17:
Check the pinned issue

closed 2020-07-17 by williambj1