OpenIntelWireless/IntelBluetoothFirmware #121: ProBook 450 g6

tradexp, 2020-06-07:
Problem on Probook 450 g6. Here is the log. thanks
2020-06-07 20:53:15.237030+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-07 20:53:15.237615+0200 0x12a      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 设备固件上传完成
2020-06-07 20:53:15.237623+0200 0x12a      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-06-07 20:53:15.237629+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-07 20:53:15.237634+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-06-07 20:53:15.238033+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-07 20:53:15.248604+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-07 20:53:15.248616+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-06-07 20:53:15.248676+0200 0x104      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-06-07 20:53:15.248738+0200 0x19d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

mehmetyukcell, 2020-06-08:
Yes, I have same issue after windows update dual boot
[image] @zxystd

mehmetyukcell, 2020-06-08:
@tradexp I fixed bluetooth issue: Delete ssdt for USB MAP or remap usb. FİXED: @zxystd thanks for your guiding

tradexp, 2020-06-08:
Good news ! But not clear what you mean by "Delete ssdt for USB MAP or remap usb"
Should I regenerate SSDT-UIAC, SSDT-USBX + USBPorts.kext ?
I already did it with Bluetooth on HS10 USB port without success
Can you please provide more details ?  Thanks !

mehmetyukcell, 2020-06-08:
> Good news ! But not clear what you mean by "Delete ssdt for USB MAP or remap usb"
> Should I regenerate SSDT-UIAC, SSDT-USBX + USBPorts.kext ?
> I already did it with Bluetooth on HS10 USB port without success
> Can you please provide more details ? Thanks !

[image]
[image]
[image]

tradexp, 2020-06-09:
Thanks very much for the details.
But still doesn't work for me. Bluetooth is detected, but continuously scanning for devices
USB port is HS10

[image]

[image]

same log from Intel firmware kext

2020-06-09 20:59:38.330559+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-09 20:59:38.330756+0200 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 设备固件上传完成
2020-06-09 20:59:38.330763+0200 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-06-09 20:59:38.330769+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-09 20:59:38.330775+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-06-09 20:59:38.331156+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-06-09 20:59:38.341764+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-06-09 20:59:38.341774+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-06-09 20:59:38.341858+0200 0x155      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-06-09 20:59:38.341921+0200 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

tradexp, 2020-06-15:
Solved it with Mojave version of the kext from here
https://github.com/zxystd/IntelBluetoothFirmware/issues/7 (version 1.01)
So the version 1.03 is Catalina only
It would be nice to add here a Mojave version or provide procedure to recompile the source for Mojave
Thanks

zxystd, 2020-06-15:
Thank you for figure it out, I will take some time to have a look.

opzhang, 2020-07-20:
I put the Bluetooth driver in clover, and it doesn't load at all when I put it in SLE. Intel 3160 network card

closed 2021-07-04 by zxystd