OpenIntelWireless/IntelBluetoothFirmware #124: Broken Bluetooth

nvectorrr, 2020-06-13:
Disconnects everytime, no sound. Intel 7260. Trying to connect to airpods pro

nvectorrr, 2020-07-09:
<code><➜  ~ log show --last boot | grep IntelFirmware
2020-07-08 19:02:06.045008+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-08 19:02:06.162067+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-07-08 19:02:06.173068+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-07-08 19:02:06.188070+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-08 19:02:06.188811+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EXIT_MFG
2020-07-08 19:02:06.189292+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-07-08 19:02:06.201092+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-08 19:02:06.279563+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-07-08 19:02:06.280224+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-07-08 19:02:06.281755+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-07-08 19:02:16.308947+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-08 19:02:16.325891+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-07-08 19:02:16.327048+0300 0x141      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
2020-07-09 17:55:41.920703+0300 0x6f       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-07-09 17:55:41.921272+0300 0x74       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()
2020-07-09 17:55:43.434270+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-07-09 17:55:43.434291+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-07-09 17:55:43.434305+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-07-09 17:55:43.434432+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-07-09 17:55:44.618124+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-07-09 17:55:44.618715+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found interface!!!
2020-07-09 17:55:44.619102+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Interrupt endpoint!
2020-07-09 17:55:44.619302+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Bulk out endpoint!
2020-07-09 17:55:44.619474+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Bulk in endpoint!
2020-07-09 17:55:44.619645+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-07-09 17:55:44.619649+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_RESET
2020-07-09 17:55:44.621130+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-07-09 17:55:44.630954+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-09 17:55:44.630960+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-07-09 17:55:44.631278+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-07-09 17:55:44.631938+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found device firmware ibt-hw-37.7.10-fw-1.80.2.3.d.bseq
2020-07-09 17:55:44.632013+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: request firmware success
2020-07-09 17:55:44.632029+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-09 17:55:44.632033+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_ENTER_MFG
2020-07-09 17:55:44.632256+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-07-09 17:55:44.632937+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-09 17:55:44.740943+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-07-09 17:55:44.741939+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
[... cut ...]
2020-07-09 17:55:44.757942+0300 0x106      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-07-09 17:55:44.757953+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-07-09 17:55:44.757961+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-07-09 17:55:44.757963+0300 0x5cf3     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
/code>

williambj1, 2020-07-09:
please use the issue template.

williambj1, 2020-07-13:
@zxystd

closed 2020-07-17 by williambj1