OpenIntelWireless/IntelBluetoothFirmware #126: Bluetooth does not work correctly on MSI MEG X570 Unify

mweber12, 2020-06-17:
**Description**
I am using an AMD Ryzen system with a MSI MEG X570 Unify board that has, AFAIK, an Intel AX200 chip for Wifi/Bluetooth.
I got it working once, but it didn't connect to a Logitech ERGO K860 keyboard or Logitech MX Vertical mouse. I was able to connect to AirPods, but I didn't hear anything there nor did the microphone work.

Now, even bluetooth doesn't startup anymore.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel AX200 (afaik) onboard
- Product ID: 0x0029
- Manufacturer ID: 0x8087  (Intel Corporation)
- macOS Version: 10.15.5
- OpenCore Version: 0.59

**Info in System Report - Bluetooth**

```
no information
```

**Logs**
```log
2020-06-17 17:28:12.970860+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-06-17 17:28:12.988646+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-06-17 17:28:12.989664+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-06-17 17:28:12.990699+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-06-17 17:28:14.547242+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-06-17 17:28:14.586003+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-17 17:28:14.586937+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-17 17:28:14.622046+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-17 17:28:14.623051+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-06-17 17:28:14.624065+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-06-17 17:28:14.669522+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-06-17 17:28:14.728606+0200 0x214      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
```

**Kernel Panic Logs**
no kernel panic

zaptrem, 2020-06-28:
Same here, except my AX200 is an add-in-board that uses an internal USB bus for the Bluetooth portion.

mweber12, 2020-06-30:
I have more information now:

Bluetooth can be activated if I first boot into Windows. Then Bluetooth on macOS can be used as well.

But I cannot connect to a device. Here is the log that I get now:
```
2020-06-30 15:07:15.902400+0200 0x211      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-06-30 15:07:15.905565+0200 0x211      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-06-30 15:07:15.906070+0200 0x211      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-06-30 15:07:15.908115+0200 0x211      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
```

zxystd, 2020-06-30:
I have the same card, and it works fine. maybe I should update OS to 10.15.5? now I am on 10.15.0

mweber12, 2020-06-30:
Do you use a dedicated card? Maybe that would also be a difference as it is builtin into my mainboard.

zaptrem, 2020-07-01:
I have a dedicated card on 10.15.5 and it doesn't work.

parizene, 2020-07-08:
@zxystd I have X570 I AORUS PRO WIFI motherboard, bluetooth audio is working using IntelBluetoothFirmware, but have issues with microphone on AirPods, it doesn't work (sound settings doesn't show levels also hangs while switching audio devices, voice memos starts, hangs, stops with no luck). Is it a known issue? 10.15.5

williambj1, 2020-07-17:
Please read the pinned issue

closed 2020-07-17 by williambj1