OpenIntelWireless/IntelBluetoothFirmware #127: Bluetooth unavailable after Wake up from sleep

quhuy0410, 2020-06-19:
**Description**
> It happens sometimes when I sleep and wakeup, I must restart to make it works again

**Environment**
- Kext Version: Lasted version
- WiFi Card Model: Intel 8265
- macOS Version: Mojave

FlasHRender, 2020-06-19:
Have the same issue on Catalina on the same card. Will attach log next time it happens.

FlasHRender, 2020-06-19:
Here is the log. BTW bluetooth is set to internal(255) on USBPorts.kext

[log.txt](https://github.com/zxystd/IntelBluetoothFirmware/files/4803039/log.txt)

cha331, 2020-06-19:
Same card. This bug seems to exist for a long time, and even cause the system to crash during sleep. I had to disable it to ensure the stability of the system.

suzuke, 2020-06-24:
Try this one: [Archive.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4826793/Archive.zip)
I am trying to fix this bug, it seems to work fine with my build.

FlasHRender, 2020-06-24:
@suzuke Thanks will test and report.

FlasHRender, 2020-06-24:
@suzuke you are a genius. I think its fixed. Tried Sleep and wake 6 time and it worked perfectly.

quhuy0410, 2020-06-25:
> @suzuke you are a genius. I think its fixed. Tried Sleep and wake 6 time and it worked perfectly.

@suzuke @FlasHRender How did you guy fix it ? I still face the problem :(

suzuke, 2020-06-25:
> > @suzuke you are a genius. I think its fixed. Tried Sleep and wake 6 time and it worked perfectly.
> 
> @suzuke @FlasHRender How did you guy fix it ? I still face the problem :(
Do you try this kext that I build in the below comment?
https://github.com/zxystd/IntelBluetoothFirmware/issues/127#issuecomment-648935657

quhuy0410, 2020-06-25:
> > > @suzuke you are a genius. I think its fixed. Tried Sleep and wake 6 time and it worked perfectly.
> > 
> > 
> > @suzuke @FlasHRender How did you guy fix it ? I still face the problem :(
> 
> Do you try this kext that I build in the below comment?
> [#127 (comment)](https://github.com/zxystd/IntelBluetoothFirmware/issues/127#issuecomment-648935657)

Of course, I've used your kext, but it's still the same :(

quhuy0410, 2020-06-25:
> Of course, I've used your kext, but it's still the same :(
Sorry but now bluetooth doesn't work from boot...

tvhung83, 2020-06-27:
> Try this one: [Archive.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4826793/Archive.zip)
> I am trying to fix this bug, it seems to work fine with my build.

@suzuke if it fixes something, please help open a PR.

suzuke, 2020-06-27:
> @suzuke if it fixes something, please help open a PR.
sure,  if there are many more people to confirm that this build fixed the bug, I will create a PR.

FlasHRender, 2020-06-27:
@suzuke I have tested it extensively and can confirm the fix actually works. Please open a PR with the changes.

yuqZY, 2020-07-06:
咋了，这是咋了？笑死了，哈哈哈

williambj1, 2020-07-13:
Should be fixed in v1.1.1, feel free to reopen if you still experience this issue.

closed 2020-07-13 by williambj1

azhinu, 2022-02-10:
> Try this one: [Archive.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4826793/Archive.zip) I am trying to fix this bug, it seems to work fine with my build.

For some reason this version works fine on Catalina, but with v2.1 I have a same issue.

peddy22, 2022-09-05:
> > Try this one: [Archive.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4826793/Archive.zip) I am trying to fix this bug, it seems to work fine with my build.
> 
> For some reason this version works fine on Catalina, but with v2.1 I have a same issue.

I'm having the OP issue with 2.2 as well on Monterey

azhinu, 2022-09-05:
@peddy22 I am using this build on Monterey. Works fine.

RobyRew, 2022-11-27:
the same issue happens still on latest version.
Here's my EFI ([link](https://github.com/RobyRew/ASUS-FX504GE-Hackintosh_OpenCore))

maksimlenskii, 2024-03-06:
the same issue happens still on latest version.
macOS Ventura MSI MPG Z790 EDGE WIFI