OpenIntelWireless/IntelBluetoothFirmware #132: [Question] Which firmware file is for which card?

igorkulman, 2020-07-05:
I am trying to build a slim version that would just support my AC 7260 so I want to ask which of the firmwares is for which card, it is not clear from their names.

I do the same for itlwm, it is easy to do there as the firmware for my card has 7260 in the name but in this repo I cannot figure out which firmware to keep.

zxystd, 2020-07-05:
different from wifi cards, the bt firmware is not hardcoded, it is read from the card, according to the card version to decide which firmware it should be used. you can build the latest code, load it and you will see `fw_name` key in the kext property, that is the firmware name. also, you can get it from the log.

igorkulman, 2020-07-05:
Thanks, found it in the logs

```
kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware ibt-hw-37.7.10-fw-1.80.2.3.d.bseq
```

closed 2020-07-05 by igorkulman