OpenIntelWireless/IntelBluetoothFirmware #134: Previous version worked well (v1.0.3). Tried out v1.1.1 and it's a no go. (ASUS Prime X299-Deluxe II)

getdown71, 2020-07-07:
**Description**
Previous version worked well (v1.0.3). Tried out v1.1.1 and it's a no go.

**Environment**

- Kext Version: v1.1.1
- WiFi Card Model: AC-9260
https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/dual-band-wireless-ac-9260-brief.pdf
- Product ID: As above
- macOS Version: 10.15.5

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

Kext did not load, so nothing here. Can share log later.

getdown71, 2020-07-07:
a full power cycle and reboot fixed the issue. Well done team. Great work here!

closed 2020-07-07 by getdown71

tvhung83, 2020-07-07:
> a full power cycle and reboot fixed the issue.

Same here, last night I tried to compile `master` myself, worked great. This morning, I replaced with release kext, then things got funky, random hangs, etc. even after a warm reboot. A cold reboot fixed it, thank you for the solution 👍