OpenIntelWireless/IntelBluetoothFirmware #137: 驱动导致wifi 无网络

image72, 2020-07-09:
**描述**
> EFI 放入kext文件开机后 MAC/WIN10 蓝牙可以使用，
MAC 下WI-FI列表始终只能找到一个网络，连接后无法连接到internet, ping or telnet. 
几次wireless diagnostics 导致直接系统重启
WIN10 WI-FI无法启用

**环境**

- Kext 版本: 1.1.1
- WiFi Card 型号:  BCM94360CS2
- 产品 ID: 
- macOS 版本: 10.15.5

**系统报告 - 蓝牙**
> 把所有内容都粘贴到下方的两个 ``` 之间

```
![image](https://user-images.githubusercontent.com/243154/87051029-a1aee080-c231-11ea-9ec0-700356446772.png)

```

**日志**
> 终端运行 `log show --last boot | grep IntelFir`，将得到的内容粘贴至 ```log 和 ``` 之间
空  未返回任何信息
```log

```

**内核崩溃日志**
> 如果出现了内核崩溃, 将 `keepsyms=1` 加入到引导参数中，然后粘贴崩溃日志到下方的两个 ``` 之间

```

```

**截图**
> 如果有必要，请在下方附加截图

williambj1, 2020-07-09:
博通网卡与此驱动无关

closed 2020-07-09 by williambj1

image72, 2020-07-09:
蓝牙还是NUC8 版载蓝牙啊

image72, 2020-07-09:
删除放进EFI中的2个kext文件后 WI-FI功能就恢复了

image72, 2020-07-09:
@williambj1 
蓝牙还是Intel NUC8 版载蓝牙
删除放进EFI中的2个kext文件后 WI-FI功能就恢复了

麻烦看一下

zxystd, 2020-07-10:
> @williambj1
> 蓝牙还是Intel NUC8 版载蓝牙
> 删除放进EFI中的2个kext文件后 WI-FI功能就恢复了
> 
> 麻烦看一下

你试试删除两个kext，然后进windows热重启回来mac，再看看WiFi正常不正常