OpenIntelWireless/IntelBluetoothFirmware #138: Intel AC 7265 not working

Xpl0itU, 2020-07-14:
**Description**
> A clear and concise description of what the bug is
I have an Intel Dual Band Wireless-AC 7265 and the kext isn't working
**Environment**

- Kext Version: 1.1.1
- WiFi Card Model: AC 7265
- Product ID: 0x095A (only the wifi one)
- macOS Version: 10.15.5

**Info in System Report - Bluetooth**

```
No information found.
```

**Logs**

```log
-blank-
```

**Kernel Panic Logs**

```
none
```

**Screenshots**
none

williambj1, 2020-07-14:
Please check your EFI configuration

Xpl0itU, 2020-07-15:
It’s this one: https://github.com/Xpl0itU/hackintosh-hp-15-bs026ns-opencore
(But with the kexts obviously)

williambj1, 2020-07-15:
So you mean the kext is working but your card isn't? Then what's the purpose of this issue? Please correct me if I'm wrong.

Xpl0itU, 2020-07-15:
No, the card works fine in other os’s, but I can’t get it to work in macOS, maybe there is something in my efi blocking it, which I doubt it

Xpl0itU, 2020-07-17:
just tried usb mapping with no results

mbirth, 2020-07-20:
I've got a HP Spectre x360 13-4104ng which also has the AC7265 and for me, Bluetooth is working fine without this kext for some reason. BT is detected via ACPI as a "Bluetooth USB Host Controller". Just can't turn it off/on from the menu.

Xpl0itU, 2020-07-20:
ok, it's working now, had to boot from a ubuntu live usb and disable and enable bluetooth

closed 2020-07-20 by Xpl0itU