OpenIntelWireless/IntelBluetoothFirmware #139: Range of bluetooth is too low

guaychou, 2020-07-15:
**Description**
> i connect to the my bluetooth. There is no sound in my speaker if range between two device more than 1 meter 

**Environment**

- Kext Version: 
- WiFi Card Model: AC7260
- Product ID: 0x087 , 0x07DC
- macOS Version: Mojave 10.14.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version:	6.0.14d3
  Hardware, Features, and Settings:
  Name:	Kevin’s MacBook Pro
  Address:	E8-2A-EA-0D-1A-32
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	6.1280 (6.1280)
  Bluetooth Power:	On
  Discoverable:	On
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x07DC
  HCI Version:	4.0 (0x6)
  HCI Revision:	0x500
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x500
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

williambj1, 2020-07-15:
Check your antenna, the range issue has nothing to do with this kext.

closed 2020-07-15 by williambj1

shreyans1, 2021-05-26:
Hi I am facing same. I did not faced same on windows and Linux