OpenIntelWireless/IntelBluetoothFirmware #142: [Validate] IORegistry fw_name empty; Info.plist has spaces before model; Hackintool unknown vendor

mackonsti, 2020-07-18:
**Description**
A few minor issues to assess and improve:
a) In IORegistry, the string name **fw_name** is empty; we can put the FW version uploaded
b) The IntelBluetoothFirmware.kext main **Info.plist** contains a space character before models in **IntelBluetoothFirmware_** keys
c) Hackintool reports vendor name as ? and **FW Loaded** not detected despite being uploaded (10.256)

**Environment**
- Kext Version: v1.1.1 release
- WiFi Card Model: Intel 9560 embedded Bluetooth 5.1 controller
- Vendor/Product ID: x8087 and 0x0aaa
- macOS Version: Mojave 10.14.6 with latest Security Update

**Screenshots**
[image]
[image]
[image]

**Logs**
As both IntelBluetooth kexts were placed in CLOVER/kext/Other folder, I cannot get any kernel info for some reason, using `log show --last boot | grep -i IntelFirmware` or any other combination :-(

williambj1, 2020-07-19:
@mackonsti Thank you very much for your detailed report, highly appreciated.

> In IORegistry, the string name fw_name is empty

This is reproducible after a sleep-wake cycle, unknown bug.

[image]

[image]

> The main firmware kext's Info.plist contains a space character before models in IntelBluetoothFirmware_ keys;

Confirmed.

> Hackintool reports vendor name as ? and FW Loaded not detected despite being uploaded;

That's because Hackintool currently doesn't support IntelBluetoothFirmware

https://github.com/headkaze/Hackintool/blob/f4a4b75c8b991d5f6456855e37a838641a65aa45/Hackintool/AppDelegate.m#L1897-L2040

I've tried to tweak that function after you created this issue, but I failed to get it working. You could create a feature request for Headkaze.

> Kextless IntelBluetoothInjector.kext v1.1.1 contains folder _CodeSignature with multiple files inside, not sure which is needed and which is there by mistake;

Xcode creates that automatically, it doesn't affect the kext's functionality.

mackonsti, 2020-07-19:
Hi @williambj1 I have contacted **headkaze** to add support for this kext/driver, and he replied the following in the meantime:

_There must be a "FirmwareLoaded" entry for the device's IOUSBDevice entry in IORegistry for Hackintool to detect a successful FW load._

[image]

Please kindly add this IORegistry entry if the FW has been successfully loaded?

headkaze, 2020-07-19:
I've got an update for Hackintool ready to test as soon as the above is added.

headkaze, 2020-07-20:
Please try [Hackintool v3.4.3](https://github.com/headkaze/Hackintool/releases/tag/3.4.3) and let me know if it shows "FW Loaded" correctly.

zxystd, 2020-07-20:
[image]
Nice support! thanks.

mackonsti, 2020-07-20:
Thanks for your fast action @headkaze ! Should we also see vendor-id 8087 as Intel Corporation here too?  (just taking the opportunity to ask before you deploy).

headkaze, 2020-07-20:
Yeah it should be showing up as Intel. I will have a look soon.

headkaze, 2020-07-20:
Can you please try [Hackintool v3.4.4](https://github.com/headkaze/Hackintool/releases/tag/3.4.4).

mackonsti, 2020-07-20:
Hi @headkaze I am sorry but I have myself no access to pre-release kext to test the FirmwareLoaded flag (I cannot build it either) but if @zxystd made it work, I have no reason to say it's not working, so I will test it when the guys release officially v1.1.2 here.

However, both the BT and WLAN kexts by OpenIntelWireless **are** detected **and** the name is fixed in BTLE device (vendor 0x8087).

(there are some minor text fixes needed, I will contact you via PM over at InsanelyMac to allow me if you want partial Git access to the strings files so with your approval, I can improve them a little e.g. for USB device vendor, change to "Intel Corporation" as it's like that in PCIID; equally to "Broadcom Inc." instead of "corp" etc. etc. really minor)

commit 1a9627f3 (2020-07-20): fix #142. support FirmwareLoaded check, thanks to @headkaze.
commit 6fd74ec7 (2020-09-23): fix #142. support FirmwareLoaded check, thanks to @headkaze.
closed 2020-09-23 by williambj1