OpenIntelWireless/IntelBluetoothFirmware #144: start fail, cannot open device

hitlcy, 2020-07-24:
Bluetooth does not work. here is the log. 

2020-07-24 16:04:25.783262-0700 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-07-24 16:04:25.783745-0700 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-07-24 16:04:25.788082-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-07-24 16:04:25.788096-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-07-24 16:04:25.788109-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-07-24 16:04:25.788209-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-07-24 16:04:27.289449-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-07-24 16:04:27.290167-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-07-24 16:04:27.290218-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-07-24 16:04:27.290466-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-07-24 16:04:27.290696-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-07-24 16:04:27.290916-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-07-24 16:04:27.290920-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-07-24 16:04:27.291608-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-24 16:04:27.302890-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.302902-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-07-24 16:04:27.303188-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-24 16:04:27.303850-0700 0x139      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware ibt-hw-37.7.10-fw-1.80.2.3.d.bseq
2020-07-24 16:04:27.303885-0700 0x139      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: request firmware success
2020-07-24 16:04:27.303897-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.303900-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_ENTER_MFG
2020-07-24 16:04:27.304173-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-24 16:04:27.304850-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.505858-0700 0x139      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-07-24 16:04:27.506865-0700 0x139      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-07-24 16:04:27.508840-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.508845-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EXIT_MFG
2020-07-24 16:04:27.509126-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-24 16:04:27.521887-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.521898-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-07-24 16:04:27.522266-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-07-24 16:04:27.522881-0700 0x139      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-07-24 16:04:27.522908-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-07-24 16:04:27.522926-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-07-24 16:04:27.522933-0700 0x7eb      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-07-24 16:04:33.229501-0700 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-07-24 16:04:33.229728-0700 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-07-24 16:06:35.062840-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-07-24 16:06:35.062852-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-07-24 16:06:35.062862-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-07-24 16:06:35.062947-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-07-24 16:06:36.564011-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-07-24 16:06:36.564037-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-07-24 16:06:36.564101-0700 0x127e     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-07-24 16:06:36.564228-0700 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-07-24 16:06:37.983180-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-07-24 16:06:37.983194-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-07-24 16:06:37.983203-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-07-24 16:06:37.983285-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-07-24 16:06:39.484077-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-07-24 16:06:39.484089-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-07-24 16:06:39.484123-0700 0x128f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-07-24 16:06:39.484311-0700 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
[... cut ...]
2020-07-24 16:10:15.302154-0700 0x16be     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-07-24 16:10:15.302166-0700 0x16be     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-07-24 16:10:15.302199-0700 0x16be     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-07-24 16:10:15.302395-0700 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()

hitlcy, 2020-07-24:
I'm on latest release 1.1.1

zxystd, 2020-10-30:
@hitlcy  EFI related issue I think. Or you can try remove IntelBluetoothInjector.kext. Thanks.

closed 2021-06-22 by zxystd