OpenIntelWireless/IntelBluetoothFirmware #145: AX201  0x8087, 0x0026  蓝牙，休眠唤醒后，无法使用

successc, 2020-07-25:
**Description**
> AX201  0x8087, 0x0026  蓝牙，休眠唤醒后，无法使用

另外请问一下，怎么编译 1.1.2

谢谢
**Environment**

- Kext Version: 1.1.1 
- WiFi Card Model: AX201
- Product ID: 0x8087, 0x0026
- macOS Version: 10.15.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

williambj1, 2020-07-25:
请填完剩下的信息

successc, 2020-07-26:
都填完了

successc, 2020-07-26:
[image]
编译出来后，只有这一个kext,  麻烦指导一下，谢谢

williambj1, 2020-07-26:
scheme 选择错误，用 kextstat 判断开机后有没有被引导正常注入

建议移步 Q 群交流，看起来不像是驱动的 bug

successc, 2020-07-26:
睡眠唤醒后，是这样的，蓝牙电源显示关闭
我已经编译出来了1.1.2，，试了还是不行
QQ群是多少呀，谢谢  

Apple蓝牙软件版本：	7.0.6f7
  硬件、功能和设置：
  名称：	Alans-iMac
  地址：	3C-58-C2-6E-D4-66
  支持低功耗蓝牙：	是
  支持“接力”：	是
  支持智能热点：	是
  生产企业：	Intel
  传输：	USB
  固件版本：	10.256 (10.256)
  蓝牙电源：	关闭
  可被发现：	打开
  可连接：	是
  自动寻找点：	关闭
  远程唤醒：	关闭
  供应商ID：	0x8087
  产品ID：	0x0026
  蓝牙核心规范：	Unknown (0xA)
  HCI修正版：	0x100
  LMP版本：	Unknown (0xA)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Desktop
  复合类设备：	0x380104
  设备类（主要）：	0x01
  设备类（次要）：	0x01
  服务类：	0x1C0
  自动寻找键盘：	关闭
  设备（已配对、已配置等）：
  Alan的AirPods Pro：
  地址：	40-70-F5-D5-EC-4D
  主要类型：	Miscellaneous
  次要类型：	Unknown
  服务：	
  已配对：	是
  已配置：	是
  已连接：	否
  蓝牙核心规范：	1.0b
  供应商ID：	0x004C
  产品ID：	0x200E
  设备类：	0x00 0x00 0x0000
  Bose Mini II SE SoundLink：
  地址：	4C-87-5D-C7-DC-F8
  主要类型：	Audio
  次要类型：	Loudspeaker
  服务：	Hands-Free unit
  已配对：	是
  已配置：	是
  已连接：	否
  生产企业：	Cambridge Silicon Radio (0x6, 0x2B45)
  蓝牙核心规范：	4.0
  固件版本：	0x0106
  供应商ID：	0x009E
  产品ID：	0x4035
  设备类：	0x04 0x05 0x240414
  支持EDR：	是
  支持eSCO：	是
  支持SSP：	是
  Wendy's iPhone：
  地址：	90-8C-43-CD-88-0F
  主要类型：	Phone
  次要类型：	Smartphone
  服务：	Phonebook, GATT, PAN Network Access Profile, AVRCP Device, Wireless iAP v2, Audio Source, Handsfree Gateway, AVRCP Device, MAP MAS-iOS, Wireless iAP
  已配对：	是
  已配置：	是
  已连接：	否
  生产企业：	Broadcom (0x9, 0x4228)
  蓝牙核心规范：	5.0
  固件版本：	0x0D60
  供应商ID：	0x004C
  产品ID：	0x7406
  设备类：	0x02 0x03 0x7A020C
  支持EDR：	是
  支持eSCO：	是
  支持SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM通道：	3
  需要认证：	否
  传出串口：
  WendysiPhone-WirelessiA：
  地址：	90-8C-43-CD-88-0F
  RFCOMM通道：	1
  需要认证：	否

successc, 2020-07-27:
已经OK了，是系统的问题，，我清理了一下缓存，就好了，，我刚换过主板，，之前用的蓝牙是BCM的

closed 2020-07-27 by williambj1