OpenIntelWireless/IntelBluetoothFirmware #146: intel 0x8087 0x0AAA - can connect from device (iphone) but not to devices (mouse)

Walterfilms, 2020-07-26:
**Description**
> A clear and concise description of what the bug is
Hello and first thanks for your work !

so here is my case : 

after installing kext, update plist, map USB, i have my bluetooth card recognized in system report and i can see my devices in system preferences/bluetooth, 

1 - i click connect, 
2 - it says connected for a few seconds, but the "wheel" is still turning
3 - the device is available to connect
4 - go back to step 1 

i tried with latest magic mouse and logitech mouse MX master S2 and  with the same issue.

sometimes i see the same device multiple times but none of them connects.

i then tried to connect from my iphone to my hackintosh and it worked, they are currently connected.

**Environment**

- Kext Version: 1.1.1
- WiFi Card Model: 0x8087
- Product ID: 0x0AAA
- macOS Version: 10.15.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Version du logiciel Bluetooth Apple :	7.0.6f7
  Matériel, fonctionnalités et réglages :
  Nom :	postprod-hack-1
  Adresse (URL) :	80-32-53-F2-C6-EF
  Bluetooth faible énergie pris en charge :	Oui
  Handoff pris en charge :	Non
  Instant Hotspot pris en charge :	Non
  Fabricant :	Intel
  Transport :	USB
  Version du programme interne :	10.256 (10.256)
  Alimentation Bluetooth :	Activé
  Détectable :	Désactivé
  Connectable :	Oui
  Rech. autom. du pointeur :	Activé
  Réactivation distante :	Activé
  ID du fournisseur :	0x8087
  Identifiant du produit :	0x0AAA
  Principales caractéristiques Bluetooth :	Unknown (0xA)
  Révision HCI :	0x100
  Version LMP :	Unknown (0xA)
  Sous-version LMP :	0x100
  Type d’appareil (principal) :	Computer
  Type d’appareil (complet) :	Mac Desktop
  Classe d’appareils composite :	0x380104
  Classe d’appareils (principale) :	0x01
  Classe d’appareils (secondaire) :	0x01
  Classe de services :	0x1C0
  Rech. autom. du clavier :	Activé
  Appareils (jumelés, configurés, etc.) :
  iPhone de pv :
  Adresse :	30-57-14-34-65-E0
  Type majeur :	Phone
  Type mineur :	Smartphone
  Services :	Phonebook, GATT, PAN Network Access Profile, Wireless iAP v2, AVRCP Device, AVRCP Device, Audio Source, Handsfree Gateway, MAP MAS-iOS, Wireless iAP
  Jumelé :	Oui
  Configuré :	Oui
  Connecté :	Non
  Fabricant :	Broadcom (0x9, 0x420E)
  Principales caractéristiques Bluetooth :	5,0
  Version du programme interne :	0x0D40
  ID du fournisseur :	0x004C
  Identifiant du produit :	0x7238
  Classe d’appareils :	0x02 0x03 0x7A020C
  EDR pris en charge :	Oui
  eSCO pris en charge :	Oui
  SSP pris en charge :	Oui
  Services :
  Transfert de fichiers Bluetooth :
  Dossier de recherche de ceux-ci :	~/Public
  Lors de la réception d’éléments :	Tout accepter sans avertissement
  État :	Désactivé
  Échange de fichiers Bluetooth :
  Dossier des éléments acceptés :	~/Downloads
  Si d’autres éléments sont acceptés :	Enregistrer dans l’emplacement
  Lors de la réception d’éléments :	Tout accepter sans avertissement
  État :	Désactivé
  Partage Internet Bluetooth :
  État :	Désactivé
  Ports série entrant :
  Bluetooth-Incoming-Port :
  Canal RFCOMM :	3
  Authentification requise :	Non
  Ports série sortant :
  iPhonedepv-WirelessiAPv2 :
  Adresse (URL) :	30-57-14-34-65-E0
  Canal RFCOMM :	1
  Authentification requise :	Non
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

the command log show --last boot | grep IntelFirmware didn't work but i had to use your alternative method : 

1. add msgbuf=1048576 into boot-args
2. in terminal use sudo dmesg | grep -i IntelFir > ~/Desktop/log.txt

here is the output :

IntelFirmware: Driver init() 
IntelFirmware: Driver Probe()
IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
IntelFirmware: Driver Start()
identified disk2 as removableidentified disk2s1 as removableIntelFirmware: set configuration to 1
IntelFirmware: Found interface!!!
IntelFirmware: Found Interrupt endpoint!
IntelFirmware: Found Bulk out endpoint!
IntelFirmware: Found Bulk in endpoint!
IntelFirmware: usb init succeed
IntelFirmware: HCI_OP_INTEL_VERSION
IntelFirmware: interrupt wait
IntelFirmware: Bootloader revision 0.1 build 42 week 52 2015IntelFirmware: suspect device firmware: ibt-17-16-1.sfi
IntelFirmware: interrupt continue
IntelFirmware: HCI_OP_BOOT_PARAMS
IntelFirmware: interrupt wait
IntelFirmware: Device revision is 2
IntelFirmware: Secure boot is enabled
IntelFirmware: OTP lock is enabled
IntelFirmware: API lock is enabled
IntelFirmware: Debug lock is disabled
IntelFirmware: Minimum firmware build 1 week 10 2014
IntelFirmware: Found device firmware: ibt-17-16-1.sfi
IntelFirmware: interrupt continue
IntelFirmware: send firmware header
IntelFirmware: send firmware pkey
IntelFirmware: send firmware signature
IntelFirmware: send firmware data
IntelFirmware: boot_param=0x40800
IntelFirmware: send firmware done
IntelFirmware: interrupt continue
IntelFirmware: HCI_OP_INTEL_RESET
IntelFirmware: Intel reset succeed
IntelFirmware: Notify: Firmware download done
IntelFirmware: can not proceed response
IntelFirmware: interrupt wait
IntelFirmware: Notify: Device reboot done
IntelFirmware: can not proceed response
IntelFirmware: interrupt continue
IntelFirmware: HCI_OP_INTEL_EVENT_MASK
IntelFirmware: interrupt wait
IntelFirmware: interrupt continue
IntelFirmware: End download
IntelFirmware: Clean up...

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

Scrubber777, 2020-08-04:
I have the same problem, I cannot connect my bluetooth-mobile-mouse-3600
I never found a solution

williambj1, 2020-08-04:
Duplicate of #51

closed 2020-08-04 by williambj1