OpenIntelWireless/IntelBluetoothFirmware #149: Bluetooth does not Work on cold start but works after 1st sleep

schmosbyy, 2020-07-29:
So, once the laptop is turned on from a shutdown state, the bluetooth is not available/working as shown below:
[image]

But after the laptop is put to sleep and when it is woken up, the bluetooth starts working as shown below:
[image]

Kext version:1.1.1
OS version: 10.15.4(Catalina)
wifi version: 8265(Asus vivobook)
Kext installed at : Clover/other + Library/extensions
(Bluetooth does not work if just kept in Clover/other)
kindly help in resolving this issue. Thanks!

Batyan45, 2020-08-04:
Same problem on Mi Notebook Air 2018

orveomindgrind, 2020-10-20:
I have the same problem

williambj1, 2020-10-20:
@schmosbyy Please don't install the kext in multiple locations, I suspect you configurated kext inject in Clover as Detect so it completely ignored the presence of its other folder. The Bluetooth Kext might have loaded too early by Clover and caused such an issue, or it's a hardware power state problem caused by the BIOS. Try using OpenCore and inject the kext at the latest position.

williambj1, 2020-10-20:
@Batyan45 @orveomindgrind Please provide more evidence.

borasumer, 2020-12-16:
I have exactly the same issue. When I start it, Bluetooth is off and cannot be turned on manually, and I hit sleep, then wake it up, and magically Bluetooth is on. Not sure what the issue is here, and not sure why this is not a bug but a question. What proof do you need so I can provide ?

closed 2021-06-22 by zxystd