OpenIntelWireless/IntelBluetoothFirmware #150: intel AC9462蓝牙找不到设备

MicalShow, 2020-07-29:
**Description**
> A clear and concise description of what the bug is
The built-in Bluetooth of the intel ac9462 wireless network card cannot be recognized. BIOS and EFI problems have been eliminated. The same machine can work normally if it is replaced with AX200 and AC9560 Bluetooth. If it is replaced with AC9462, macOS cannot detect the Bluetooth device.

The log is blank and there is no output.

**Environment**

- Kext Version: 1.1.1
- WiFi Card Model: Intel AC-9462
- Product ID: 0x0002
- macOS Version: 11.0

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

zxystd, 2020-07-29:
@MicalShow  Can you confirm your device product id is 0x0002? I can't find it in linux code. would you mind upload the screenshot about the BT info?

MicalShow, 2020-07-29:
@zxystd oh，sorry，should be 0x0aaa.

zxystd, 2020-07-29:
0xaaa is supported, if it the driver not load, I can't do anything, maybe you should check your EFI.

MicalShow, 2020-07-31:
@zxystd Hi bro, my problem was successfully resolved. The cause of this problem was that the EFI I used incorrectly disabled the Bluetooth port. Thank you for your help.

closed 2020-07-31 by MicalShow