OpenIntelWireless/IntelBluetoothFirmware #151: When Heliport is turned on bluetooth automatically disconnects!!!!

Tomanight, 2020-07-30:
I am a newb in hackintoshing. It may be a minor bug but when I turn on Wifi from Heliport app the bluetooth automatically disconnects. The wifi is not even connected, just turning the wifi on from the heliport app disconnects all bluetooth connected devices. I don't know how to get logs and everything that's why I have opened a blank issue

Tomanight, 2020-07-30:
I have attached 3 screen shots 
1. Before turning wifi on from heliport
2. after turning it on
3. turning off the wifi
[image]
[image]
[image]

zxystd, 2020-08-03:
which device?
and please use 5Ghz WiFi, thanks.

Tomanight, 2020-08-03:
intel wireless 7265, yeah i have tried connecting to 5ghz wifi and then connecting to my bluetooth, but it disconnects. if I try connecting to bluetooth first and then connect to wifi, the bluetooth disconnects. its like each is connected to each other if one turns on other disconnects, and vice versa

khevin014, 2020-10-06:
Had the same issue, I can't get both working at the same time. If I include itlwm kext, bluetooth will not work but wifi connection is fine. If I don't include itlwm (completely disregarding wifi) bluetooth works fine. Tried on Catalina 10.15.7 and thought it was a problem with Catalina so I upgraded to Big Sur beta 9 and no luck. Using intel 7265 with id 0x8087, 0x0A2A.

Kext version
itlwm v1.1.0 stable with heliport v1.0.1
IntelBluetoothFirmware 1.1.2
I also tried replacing itlwm with AirportItlwm_v1.0_Beta_Catalina and Big Sur with no luck.

theroadw, 2021-03-31:
Same behavior here on AC-9560
itlwm v1.3  (also on previous builds)
Forced 5GHz on router didn't change the Bluetooth disconnecting upon enabling wifi.

Wifi works and is stable
Bluetooth works and is stable as long as wifi is off.

It seems Bluetooth range drops to 1 or 2 feet and there's massive packet loss if wifi is enabled, suggesting interference.

Enabling Bluetooth while wifi is on doesn't seem to affect wifi, but Bluetooth remains unusable

Tomanight, 2021-03-31:
Are you using Opencore? if so use Airportitlwm.kext from the same Repo. Then use both the bluetooth firmware kexts, mine started working correctly when i changed to opencore and used the correct kexts.

zxystd, 2021-07-04:
It is a Coex issue with AC7265 hardware, it should be fix in AirportItlwm/itlwm if necessary. The firmware kext works fine, so I am going to close this issue.

closed 2021-07-04 by zxystd