OpenIntelWireless/IntelBluetoothFirmware #156: Logitech M720 Bluetooth cannot connect

Jaydar, 2020-08-04:
**Description**
Logitech M720 Bluetooth cannot connect

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: AX201
- Product ID: 0x8087, 0x0026
- macOS Version: 10.15.6

Scrubber777, 2020-08-05:
after installing kext, update plist, map USB, i have my bluetooth card recognized in system report and i can see my devices in system preferences/bluetooth,

1 - i click connect,
2 - it says connected for a few seconds, but the "wheel" is still turning
3 - the device is available to connect
4 - go back to step 1

i tried with Microsoft Mobile Mouse 3600 (Bt 4.0) and Xiaomi Mi Dual Mode Wireless (Bt 4.2) and with the same issue.

sometimes i see the same device multiple times but none of them connects.

i then tried to connect from my iphone to my hackintosh and it worked, they are currently connected.

Environment

Kext Version: 1.1.1
WiFi Card Model: 8265
Bluetooth: 0x8087, 0x0a2b
macOS Version: 10.15.6
Bootloader : OpenCore 0.6.0

williambj1, 2020-08-05:
Duplicate of #51

closed 2020-08-05 by williambj1