OpenIntelWireless/IntelBluetoothFirmware #160: 连接AirPods二代时检测不到麦克风输入-Cannot detect mic input with Airpods2

Aaron2018, 2020-08-08:
**Description**
> A clear and concise description of what the bug is
连接AirPods二代时检测不到麦克风输入,但是在音频设置界面可以看到AirPods的输入选项

Cannot detect mic input with Airpods2，but it could be find in audio setting.
**Environment**
nuc8i5beh
opencore
catalina 10.15.5

- Kext Version: 
1.1.2

- WiFi Card Model: Built-in
- Product ID: Built-in
- macOS Version: 10.15.5

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
  Apple蓝牙软件版本：	7.0.6f7
  硬件、功能和设置：
  名称：	Carey的Mac mini
  地址：	0C-7A-15-C2-97-D6
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	10.256 (10.256)
  蓝牙电源：	打开
  可被发现：	关闭
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0AAA
  蓝牙核心规范：	Unknown (0xA)
  HCI修正版：	0x100
  LMP版本：	Unknown (0xA)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Desktop
  复合类设备：	0x380104
  设备类（主要）：	0x01
  设备类（次要）：	0x01
  服务类：	0x1C0
  自动寻找键盘：	打开
  设备（已配对、已配置等）：
  Carey的AirPods：
  地址：	14-87-6A-4C-7D-A1
  主要类型：	Audio
  次要类型：	Headphones
  服务：	Audio Sink, AVRCP Controller, AAP Server, Handsfree, AVRCP Target
  已配对：	是
  已配置：	是
  已连接：	是
  生产企业：	Apple (0x9, 0x7D31)
  蓝牙核心规范：	5.0
  固件版本：	2​D​15
  供应商ID：	0x004C
  产品ID：	0x200F
  设备类：	0x04 0x06 0x240418
  RSSI：	0
  角色：	主设备
  连接模式：	活跃模式
  时间间隔：	0 ms
  支持EDR：	是
  支持eSCO：	是
  支持SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM通道：	3
  需要认证：	否
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.
[image]

wveilleux, 2020-12-21:
I have the same problem with the same hardware (nuc8i5beh) but with Big Sur (11.0) and Bluetooth output is working properly with a few BT Device but the Mic Input never work. Is there a way to diagnose it and find why it is like this ?

mn19941219, 2021-01-21:
same problem with AirPods2

wveilleux, 2021-01-21:
Is there any plan to support mic with that Bluetooth kexts ?

williambj1, 2021-01-24:
This issue is not related to this kext, please read the docs for more into.

closed 2021-01-24 by williambj1