OpenIntelWireless/IntelBluetoothFirmware #161: Sudden Bluetooth not working

xNoEndx, 2020-08-09:
Without anything done from my side, my bluetooth stopped working. I have no idea what went wrong. But below might help

When I run 
`sudo kextutil -t /Volumes/EFI/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext `

I get

>Kext rejected due to improper filesystem permissions: <OSKext 0x7fba1c435df0 [0x7fff880b7cc0]> { URL = "file:///Volumes/EFI/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: not code signed
Authentication Failures: 
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other): 
        /Volumes/EFI/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext
        Contents
        Info.plist
        ._Info.plist
        MacOS
        IntelBluetoothFirmware
        ._IntelBluetoothFirmware
        ._MacOS
        ._Contents

I tried many times to 

`chown -R root:wheel`

but will always defaults back to "_unknown"

The funny thing is that I didn't do anything!!

williambj1, 2020-08-10:
Kexts injected by bootloaders don't require root permission, please describe when and how it stops working and provide logs, thanks!

xNoEndx, 2020-08-10:
@williambj1 as mentioned above, I didn't change anything. Went to bed, came next day and bluetooth is showing "Not Available". Is there any command and place to check what went wrong?

williambj1, 2020-08-10:
I think this issue is of the scope of this kext since it's only a firmware uploader and leaves all the stuff for native kexts.

Ideally, you won't even need this kext at all if you perform a warm reboot from other operating systems.

However, you could try shutting down your laptop completely or do a direct reboot from Linux, that might do the trick.

closed 2020-08-10 by williambj1

xNoEndx, 2020-08-10:
@williambj1 you are right, when I did restart my PC to windows the bluetooth was off. When I did full shut down then boot into windows it was on. But when I do the same into Mac it doesn't work again!!