OpenIntelWireless/IntelBluetoothFirmware #163: Lagging mouse pointer response

emmetgrand, 2020-08-10:
Low mouse pointer response, slow showing the move of pointer, lagging when moving.
Mouse Jetaccess comfort OM-B90G

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: 8265
- macOS Version: Catalina 10.15.3

usr-sse2, 2020-08-13:
Are you connected to 2.4 GHz WiFi?

emmetgrand, 2020-08-13:
> Are you connected to 2.4 GHz WiFi?

No via Bluetooth.

williambj1, 2020-08-13:
Using 2.4GHz Wi-Fi with Bluetooth can cause some serious interferences, is this your case?

emmetgrand, 2020-08-13:
But on Windows this mouse works perfectly.  So the matter is in kext.

> Using 2.4GHz Wi-Fi with Bluetooth can cause some serious interferences, is this your case?

usr-sse2, 2020-08-13:
It works in Windows because of coexistence modes implemented in drivers.

williambj1, 2020-08-13:
> So the matter is in kext

Not at all, like this kext's name describes, it's a firmware uploader, not a driver. All it does is uploading the firmware to your card and the rest are handled by Apple's kexts. In fact, you can use Bluetooth fine if you directly reboot from Windows without this kext.

closed 2020-08-15 by williambj1