OpenIntelWireless/IntelBluetoothFirmware #164: No conection to Magic Mousse

chris1111, 2020-08-11:
I'm using ➤ [PCE-AC55BT-B1](https://www.asus.com/ca-en/Networking/PCE-AC55BT-B1/)
No device detect ?
[image]

If I disconnect the BT cable of the motherboard and I reboot then use my ORICO - Dongle USB Bluetooth 4.0 (BTA-403-BK)

its work 
[image]

williambj1, 2020-08-11:
Duplicate of #51

chris1111, 2020-08-11:
Ok today I have replug the USB  Cable in the PCIe card then I use the 2 kexts on for Bluethoot on OC then everything works correctly
Thank you for the work ! You can delete this Issue
[image]