OpenIntelWireless/IntelBluetoothFirmware #166: Bluetooth stops working due to HCI timeouts?

zearp, 2020-08-12:
Every now and then, about once every 3-4 hours Bluetooth stops working.

- Bluetooth connections get restored after a few minutes, not always.
- Turning bluetooth off and on always fixes it. It takes a while before it's actually disabled. Enabling is fast.

**Environment**

- Using the current release found on this repo.
- WiFi Card Model: Intel 9560
- macOS Version: 10.15.6

**Info in System Report - Bluetooth**

There's not much in the logs regarding this, but right when it happens there's this entry:

```
OpCode 0x200A (LE Set Advertise Enable) from: bluetoothd (111)  Synchronous  status: 0x00 (kIOReturnSuccess) state: 2 (BUSY) timeout: 4321 originalTimeout: 4321] Bluetooth warning: An HCI Req timeout occurred.
```

Some other errors about not being able to scan are there too but don't contain any useful info. The above message repeats itself until bluetooth "heals" itself, or if it takes too long when I turn it off and on.

The errors seems (to my uneducated mind) to be about some kind of power saving. I haven't timed it but it appears to happen evert 3-4 hours. sometimes it also doesn't work after waking up from sleep but I see that an issue related to that is already open.

Is it possible to disable low energy mode stuff for the time being to see if that solves it?

**Logs**

```
Will be added after fresh reboot
```

I've disabled other kexts like *itlwm* and anything else that's not essential to make troubleshooting easier. Is there a method to collect more logs or provide better info and hopefully help solve this?

Thank you!

anker2000, 2020-08-15:
I'm having the exact same issue, even more frequently than 3-4 hours. Very happy to provide a bigger chunk of my log file if it helps.

zearp, 2020-08-16:
I'm wondering if trying different firmware versions could help stability. Apparently if you boot into Windows (maybe Linux too?) it will load a firmware that lasts between reboots, so booting Windows then rebooting back into macOS without loading this kext and bluetooth should still work. I'm curious how stable that is and which firmware version it uses.

williambj1, 2020-08-16:
Firmware files used by this kext are from Linux Firmware, you can download slightly older versions of firmware files from Intel. We have no idea which version is your Windows or Linux driver using and it usually doesn’t matter.

zearp, 2020-08-16:
Thats a bummer. How would I go about and try to sort this out? I'd love to contribute and provide some debug logs or something more than this. At the moment bluetooth is too unstable to be usable. It gets quite frustrating specially since sometimes the connections aren't restored until I turn bluetooth on and off.

Is there a way to get some better logs off what's happening? Maybe a separate log file cuz now I have to dig trough all system logs hoping to find something that could help. Feels it's so close to have stable bluetooth on Intel yet with every d/c feels so far away again.

williambj1, 2020-08-16:
If your issue exists when you reboot from other operating systems then it’s unrelated to this kext since it’s only a firmware uploader, other stuff are done by Bcrm drivers which only provide poor to medium compatibility.

zearp, 2020-08-16:
I have no problems with bluetooth cutting out in Windows or Linux.

zearp, 2020-08-16:
I have no problems in other operating systems. Thats why I mentioned that maybe they use a different firmware. If they use the same firmware then issue lies in those drivers you mentioned. If the firmwares are different, I'd love to try those firmwares with this kext. But you said it usually doesn't matter.

williambj1, 2020-08-16:
> If your issue exists when you **reboot** from other operating systems 

Please read carefully, I said **reboot**

Closing as this issue is already off-topic and can't be solved with the kext's current targeted design.

closed 2020-08-16 by williambj1

zearp, 2020-08-16:
> I said **reboot**

Wow. Thanks. I'll just use my Bluetooth dongle then that works out of the box. Why bother trying to help this project with such an attitude?

williambj1, 2020-08-16:
I'm not sure what's wrong with your attitude (changing the title to nonsense, blaming others, etc) after I kindly pointed out a point of misunderstanding. I've already explained as much as I could.

A firmware file does NOT handle all of the Bluetooth's functionality, HCI commands, coex, and other stuff are out of scope with this kext.

Nobody is forcing you to use this kext and it is not designed to be a replacement of USB Bluetooth dongles which uses chips natively supported by Apple.