OpenIntelWireless/IntelBluetoothFirmware #168: 使用后依然打不开蓝牙，Mac上蓝牙 未找到信息

quantumsquirrel, 2020-08-15:
**Description**
> 打不开设备

**Environment**

- Kext Version: Latest
- WiFi Card Model: Intel ac3160
- Product ID: 
- macOS Version: 10.14.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
未找到信息
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```l2020-08-15 23:42:10.129481+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-08-15 23:42:10.129489+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-08-15 23:42:10.129496+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-08-15 23:42:10.129531+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-08-15 23:42:11.629746+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: start fail, can not open device
2020-08-15 23:42:11.629750+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
2020-08-15 23:42:11.629759+0800 0x1dc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-08-15 23:42:11.629805+0800 0x74       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()

```

quantumsquirrel, 2020-08-15:
### Update：
2020-08-15 17:40:26.774427+0800 0x1d1      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-08-15 17:40:26.785758+0800 0x1d1      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-08-15 17:40:26.864375+0800 0x1d1      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-08-15 17:40:26.864569+0800 0x1d1      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()

williambj1, 2020-08-15:
找不到设备是 efi 问题，检查一下是不是定制 usb 的时候屏蔽了。不安装驱动 sys report 里也会有蓝牙的

closed 2020-08-15 by williambj1