OpenIntelWireless/IntelBluetoothFirmware #169: AirPods and bluetooth speakers no sound on 5ghz wifi and some stutter on 2.4ghz

redouglas39, 2020-08-15:
**Description**
I have itlwm installed and the put both IntelBluetoothFirmware and intelbluetoothinjector to the kext folder and enabled them in config.plist, the wifi works fine with both 2.4 and 5ghz but the bluetooth connection is not stable especially for 5ghz when watching any video or hearing songs. My card is AC7265, any solution? Thanks so much

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: AC7265
- Product ID: -
- macOS Version: 10.15.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features, and Settings:
  Name:	
  Address:	
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4096 (8.4096)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2A
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x1000
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1000
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
2020-08-15 18:19:42.403833+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-08-15 18:19:42.404068+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-08-15 18:19:42.408763+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-08-15 18:19:42.408957+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-08-15 18:19:52.909876+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-15 18:19:52.915817+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-08-15 18:19:52.922007+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-15 18:19:52.927712+0800 0x123      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-08-15 18:19:52.933513+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-15 18:19:52.939103+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-08-15 18:19:52.944558+0800 0x170      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

i3p9, 2020-09-28:
This is a known issue for Intel 7xxx series. I suggest getting a Intel 8xxx series card. https://github.com/OpenIntelWireless/itlwm/issues/85

Doregon, 2020-12-01:
The thing is, it happens for the AC 3165 as well.

kylegorter, 2021-06-01:
@Doregon Yeah man, happening to me too on the same wifi card. Handoff and continuity not detecting Apple devices. AirPods connect but are very unstable. They become stable only when I put both my hands on my head which is extremely weird. Any other devices I've tried like connecting to a Logitech Bluetooth mouse don't work.

Doregon, 2021-06-01:
@kylegorter My “features” are that

I can’t turn my head
I can’t put my hand on the middle of my laptop
If I pass between even *paper* it cuts out
Covering one of the AirPods also cuts out

I don’t have other BT devices, but like you Handoff is super finnicky and no other continuity features work.

By this point I ended up buying a new Broadcom WiFi card.

Doregon, 2021-06-01:
Since it’s not getting fixed any time soon. https://github.com/OpenIntelWireless/itlwm/issues/85#issuecomment-689254944

kylegorter, 2021-06-01:
@Doregon It’s so similar to me. I figured that it would take a while to get it fixed so I bought a bcm94350. What did you get?

Doregon, 2021-06-01:
BCM94352

Doregon, 2021-06-01:
@kylegorter I bought a BCM94352

closed 2021-06-02 by williambj1