OpenIntelWireless/IntelBluetoothFirmware #170: Bluetooth doesn't work on Big Sur+OpenCore

GGorAA, 2020-08-16:
**Description**
I can't use Bluetooth on macOS Big Sur with OpenCore. Both kexts are launched(maybe, I injected them in config.plist), but nothing is working
**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: Intel Wireless AC 8260
- Product ID: idk what this means
- macOS Version: Big Sur Public Beta 1
[image]

**Info in System Report - Bluetooth**
[image]

```
  Apple Bluetooth Software Version:	8.0.0d44
  Hardware, Features and Settings:
  Name:	MacBook Pro — Егор
  Address:	34-F3-9A-42-67-E6
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	7.256 (7.256)
  Bluetooth Power:	On
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.1 (0x7)
  HCI Revision:	0x100
  LMP Version:	4.1 (0x7)
  LMP Subversion:	0x100
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```

**Logs**

```log
nothing in the output
```

**Screenshots**
[This](https://github.com/GGorAA/Hackintosh-840/blob/master/IntelWireless/EFI/OC/config.plist#L710) is a line in config.plist for OpenCore that enables IntelBluetoothFirmware.kext

[This](https://github.com/GGorAA/Hackintosh-840/blob/master/IntelWireless/EFI/OC/config.plist#L684) is where I enabled IntelBluetoothInjector.kext

[This](https://github.com/GGorAA/Hackintosh-840) is my whole EFI

williambj1, 2020-08-16:
Did it work on previous versions of macOS?

I'm using pb1 and everything works fine.

GGorAA, 2020-08-16:
Yes, it worked with previous versions, when I was using Clover and Catalina. When I switched to OpenCore, it stopped working. Same with Big Sur

williambj1, 2020-08-16:
Unfortunately I don’t have time to investigate your EFI and we are not responsible for such issues. You can choose other sources for personal help.

closed 2020-08-16 by williambj1

xNoEndx, 2020-09-28:
@GGorAA were you able to resolve IntelBluetoothFirmware.kext with OC?

GGorAA, 2020-09-28:
Yes indeed, I resolved. Works flawlessly :D

Kaijun, 2020-10-04:
> Yes indeed, I resolved. Works flawlessly :D

how did you make it work? 

i have the same issue, no logs output completely

GGorAA, 2020-10-19:
I just forgot to enable it in my config lol

zoonoxe, 2020-12-03:
> I just forgot to enable it in my config lol

In your screenshots you already had the 3 kext enabled (intelBluetoothFirmware, intelBluetoothInjector and itlwm). 
I have that problem, these 3 kexts enabled, the wifi works but bluetooth is not detected with Opencore. With Clover it's detected.

Sebadier, 2020-12-07:
> > I just forgot to enable it in my config lol
> 
> In your screenshots you already had the 3 kext enabled (intelBluetoothFirmware, intelBluetoothInjector and itlwm).
> I have that problem, these 3 kexts enabled, the wifi works but bluetooth is not detected with Opencore. With Clover it's detected.

Is think this is a problem with the latest opencore. Are you using version 6.3? In OC6.2 it was working with these three kexts for me. In 6.3 bluetooth stopped working.

zoonoxe, 2020-12-07:
> Is think this is a problem with the latest opencore. Are you using version 6.3? In OC6.2 it was working with these three kexts for me. In 6.3 bluetooth stopped working.

I'm using intel 7260 with opencore 0.6.0

cartoleong, 2021-02-04:
> I'm using intel 7260 with opencore 0.6.0

Do you has been slove the problem?i have the same problem yet, system report do not detected the bluetooth device with opencore config was all enbale, please kindly help to share you results many big thanks!

yingw, 2021-03-14:
Same here. Any advise? @GGorAA Im using OC 0.6.7 + Catalina, and all kexts enabled. I guesee it's sth with OC.

Sebadier, 2021-03-14:
I solved the problem by playing with the order of the kexts.
For me the solution was to put the USBPorts.kext (or whatever you use to map the USB ports) before the IntelBluetoothFirmware.kext and the IntelBluetoothInjector.kext
So try experimenting with the kext order. I'm sure there will be a solution for you.

luopuiki, 2021-03-26:
> I solved the problem by playing with the order of the kexts.
> For me the solution was to put the USBPorts.kext (or whatever you use to map the USB ports) before the IntelBluetoothFirmware.kext and [...]

please give more information, like oc version. If possbile, please send you efi folder

Greenlandgreenrevolution, 2022-08-19:
Hello guys i m on big sur using tplink ub400 dongle but not working just show me connected but not working what should i do ?