OpenIntelWireless/IntelBluetoothFirmware #176: Bluetooth mic not working

Parthvsquare, 2020-08-20:
**Description**
> First i thought i injected wrong LayoutID
> after testing reaming, I found mic problem was only on Bluetooth devices

**Environment**

- Kext Version: Latest
- WiFi Card Model: 8265
- Product ID: 
- macOS Version: latest
[image]

**Info in System Report - Bluetooth**

```  Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features and Settings:
  Name:	MacBook Pro
  Address:	14-4F-8A-7B-BB-63
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.256 (8.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off
  Devices (Paired, Configured, etc.):
  HD 4.50BTNC:
  Address:	00-16-94-3D-A4-4E
  Major Type:	Audio
  Minor Type:	Headset
  Services:	Headset, Hands-Free unit
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Cambridge Silicon Radio (0x6, 0x21C8)
  Bluetooth Core Spec:	4.0
  Vendor ID:	0x0A12
  Product ID:	0x0001
  Class of Device:	0x04 0x01 0x240404
  RSSI:	0
  Role:	Master
  Connection Mode:	Sniff Mode
  Interval:	506.25 ms
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Keyboard K480:
  Address:	34-88-5D-DA-8A-6C
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Keyboard K480                  , Keyboard K480                  , Keyboard K480                  
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x2304)
  Bluetooth Core Spec:	3.0
  Firmware Version:	0x2803
  Vendor ID:	0x046D
  Product ID:	0xB33D
  Class of Device:	0x05 0x10 0x0540
  AFH:	Off
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	Yes
  OnePlus 7:
  Address:	98-09-CF-AB-14-17
  Major Type:	Phone
  Minor Type:	Smartphone
  Services:	OBEX Object Push, OBEX Phonebook Access Server, Android Network User, Advanced Audio, Handsfree Gateway, SIM Access, SMS/MMS, Android Network Access Point, AV Remote Control Target, Headset Gateway
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Unknown (0x9, 0x2BE)
  Bluetooth Core Spec:	5.0
  Firmware Version:	0x1436
  Vendor ID:	0x001D
  Product ID:	0x1200
  Class of Device:	0x02 0x03 0x5A020C
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Parth’s iPad:
  Address:	00-5B-94-F2-FB-75
  Major Type:	Computer
  Minor Type:	Handheld
  Services:	Handsfree Gateway, Wireless iAP v2, Audio Source, AVRCP Device, Wireless iAP, MAP MAS-iOS, AVRCP Device, GATT
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Broadcom (0x9, 0x4307)
  Bluetooth Core Spec:	5.0
  Firmware Version:	0x0D40
  Vendor ID:	0x004C
  Product ID:	0x7244
  Class of Device:	0x01 0x04 0x6A0110
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
  Outgoing Serial Ports:
  ParthsiPad-WirelessiAPv2:
  Address:	00-5B-94-F2-FB-75
  RFCOMM Channel:	1
  Requires Authentication:	No

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

``` not getting any logs
```

**Kernel Panic Logs**
> no kernel panics

**Screenshots**
> I don't think it's relevant or not
[image]

LeaveNhA, 2020-09-09:
Same problem, same outputs. OMG.

Anyone else who suffers like this ?

guillevalin, 2020-09-15:
Here I have the same issue, I paired a Bose QC 35 which works perfect for audio, but for some reason using as a microphone it kind of "crash".

My device is a Wi-Fi 6 AX201 CNVI 0x8086 0x0074 (the bluetooth ids are 0x8087 0x0026).

Bluetooth info:
  Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features, and Settings:
  Name:	Mac Pro
  Address:	XX-XX-XX-XX-XX-XX
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0026
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
Bose QC35 II:
  Address:	XX-XX-XX-XX-XX-XX
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	SPP Dev, AudioWear, Wireless iAP, AudioWear Voice, Hands-Free unit, SPP Dev, Headset
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Cambridge Silicon Radio (0x8, 0x328D)
  Bluetooth Core Spec:	4,2
  Firmware Version:	0x0201
  Vendor ID:	0x009E
  Product ID:	0x4020
  Class of Device:	0x00 0x00 0x0000
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
  Outgoing Serial Ports:
  GuillermosBoseQC35II-SP:
  Address:	XX-XX-XX-XX-XX-XX
  RFCOMM Channel:	8
  Requires Authentication:	No

benvp, 2020-09-16:
Same here... Looks like I have the same configuration. Also no output on "IntelFirmware" in boot logs.

```
  Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features and Settings:
  Name:	benRig
  Address:	58-96-1D-56-F8-87
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0026
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  benMouse:
  Address:	84-38-35-29-23-3A
  Major Type:	Peripheral
  Minor Type:	Mouse
  Services:	Apple Magic Mouse
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Apple (0x3, 0x31C)
  Battery Level:	100%
  Bluetooth Core Spec:	2,0
  Firmware Version:	0x0306
  Vendor ID:	0x05AC
  Product ID:	0x030D
  Class of Device:	0x05 0x20 0x2580
  AFH:	On
  AFH Map:	3033FFC2FFF0000
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	Yes
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	No
  benKeyboard:
  Address:	04-69-F8-CD-39-B9
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Magic Keyboard
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Broadcom (0x5, 0x240C)
  Bluetooth Core Spec:	3,0
  Firmware Version:	0x0100
  Vendor ID:	0x004C
  Product ID:	0x0267
  Class of Device:	0x05 0x10 0x2540
  Host Connectable:	Yes
  EDR Supported:	Yes
  eSCO Supported:	No
  SSP Supported:	Yes
  Boss’s AirPods Pro:
  Address:	14-87-6A-13-3A-0A
  Major Type:	Audio
  Minor Type:	Headphones
  Services:	Audio Sink, AVRCP Controller, AAP Server, Handsfree, AVRCP Target
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Manufacturer:	Apple (0x9, 0x7D31)
  Bluetooth Core Spec:	5,0
  Firmware Version:	2​D​27
  Vendor ID:	0x004C
  Product ID:	0x200E
  Class of Device:	0x04 0x06 0x240418
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

MoSchaub, 2020-09-29:
Same Problem with AirPods on Intel Nuc8i7BEH no mic but sound works perfectly fine

closed 2020-10-07 by Parthvsquare

jovan90jovanovic, 2020-12-16:
I have the same problem with AirPods Mic. Can be selected but when selected, bugs all other bluetooth connections and doesn't work. Audio Output works fine. Why wontfix?

fumpierrez, 2020-12-25:
Having same issue, headphones audio works great... but stops working when I try to use the microphone (audio stops, microphone doesn't work)

aidanchandra, 2021-03-16:
@MoSchaub sorry to post on a dead threat but I think it'd be less annoying than opening a new one.
Just curious why this got the nofix label. Is it certainly an issue with AppleALC and NOT Bluetooth?

richibrics, 2021-05-06:
Same problem with AirPods and IntelBluetooth, output works fine but no microphone sound and if I try to select it in the settings, the window become unresponsive

jamesst20, 2021-05-19:
@williambj1 

May I kindly ask why it's labeled as wontfix? I'm also suffering from the same issue with Sony bluetooth headset on AX1650. Is it the lack of information or really because it's caused by something else? Could we get some tip?

williambj1, 2021-05-20:
@jamesst20 

It’s labeled as won’t fix because currently we have no idea what is causing this issue (maybe timing) and @zxystd has no plan to investigate further. A completely new project based on Lilu patches or completely writing a new HostController kext, which both requires extensive reverse engineering, is needed to make Bluetooth work better.

#262 is a same issue report as this one, not only contains a much detailed description but also a workaround.

jamesst20, 2021-05-20:
> @jamesst20
>
> It’s labeled as won’t fix because currently we have no idea what is causing this issue (maybe timing) and @zxystd has no plan to investigate further. A completely new project based on [...]

Thanks for the quick reply! Indeed I had noticed it was fine when rebooting from Windows. Are there any permanent work around? What if I were to get a USB Bluetooth adapter?

williambj1, 2021-05-20:
> > @jamesst20
> > It’s labeled as won’t fix because currently we have no idea what is causing this issue (maybe timing) and @zxystd has no plan to investigate further. A completely new project based on Lilu patches or completely writing a new HostController kext, which both requires extensive reverse engineering, is needed to make Bluetooth work better.
> > #262 is a same issue report as this one, not only contains a much detailed description but also a workaround.
> 
> Thanks for the quick reply! Indeed I had noticed it was fine when rebooting from Windows. Are there any permanent work around? What if I were to get a USB Bluetooth adapter? Would it work and would it conflict with my internal bluetooth?

Currently we don’t know other workarounds. Not sure what will happen if there are two Bluetooth adapters, but you can always disable IntelBluetooth easily by excluding that USB port from your USB mapping configs.

jamesst20, 2021-05-20:
> Currently we don’t know other workarounds. Not sure what will happen if there are two Bluetooth adapters, but you can always disable IntelBluetooth easily by excluding that USB port from your USB [...]

100% right ! I will try to find on Amazon a compatible bluetooth adapter and let you know how it goes when it's delivered. Expect it to take a week

jamesst20, 2021-05-27:
Leaving my answer here too to make it more visible

> @williambj1
> 
> Just letting you know. I got the ASUS USB BT-400 (Broadcom based) and it suffers from the same audio input issue. System Preference --> Sound slow to open and the input bar is not moving when I speak
> 
> [image]
> 
> Edit:
> 
> I also tried removing intel bluetooth from USB mapping and also removed the Intel kext. Same result.
> 
> What does it mean to us? The ASUS BT-400 is natively supported, I don't need any kext. (I'm using Big Sur 11.2.3). The issue is not specific to intel or this kext

richibrics, 2021-05-27:
With my AirPods and my internal Bluetooth controller I have the same microphone problem, but I tried with an old USB Bluetooth controller and the AirPods microphone works.

Parthvsquare, 2021-05-27:
> With my AirPods and my internal Bluetooth controller I have the same microphone problem, but I tried with an old USB Bluetooth controller and the AirPods microphone works.

which one?
is that airpod normal or pro's. Also could you check the wifi as the intel one will suffer if internal bluetooth is being used