OpenIntelWireless/IntelBluetoothFirmware #177: the kext is basically useless if used with itlwm

tommyvct, 2020-08-26:
**Description**
the kext is basically useless if used with `itlwm`. Will not inject the firmware at all if not previously rebooted from Windows.

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: AC3168
- Product ID: 
- macOS Version: 10.15.6

**Info in System Report - Bluetooth**
```
No information found.
```

**Logs**
Sometimes the message between the Interrupt differs.

```log
2020-08-26 02:05:28.816465-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-08-26 02:05:28.823511-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-08-26 02:05:28.825630-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-08-26 02:05:28.830076-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-08-26 02:05:28.834426-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-08-26 02:05:28.838835-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-08-26 02:05:28.840846-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:05:28.845184-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:05:38.847330-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:05:38.847959-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:05:38.850842-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:05:49.796527-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:05:49.797910-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:05:49.801623-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:05:59.804015-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:05:59.805484-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:05:59.809414-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:09.812110-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:09.813569-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:09.817956-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:19.820404-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:19.821890-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:19.825627-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:29.827236-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:29.828663-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:29.832261-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:39.833983-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:39.833987-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:39.836953-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:49.837059-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:49.837063-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:49.839590-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:06:59.839905-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:06:59.839909-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:06:59.842260-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:09.843286-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:07:09.843290-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:07:09.845917-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:19.846155-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:07:19.846159-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:07:19.848589-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:29.849210-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:07:29.849214-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:07:29.852232-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:39.853234-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:07:39.853239-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:07:39.855894-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:49.857207-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:07:49.857212-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:07:49.859567-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:07:59.860610-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
[... 279 lines cut ...]
2020-08-26 02:23:30.168283-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:23:30.170931-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
[... cut ...]
2020-08-26 02:24:30.189271-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:24:30.189274-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:24:30.191904-0500 0x2e9      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait

```

**Kernel Panic Logs**
no kernel panic caused by this.

tommyvct, 2020-08-26:
```
2020-08-26 02:47:14.133927-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-08-26 02:47:14.140957-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-08-26 02:47:14.143144-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-08-26 02:47:14.147563-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-08-26 02:47:14.152294-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-08-26 02:47:14.156623-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-08-26 02:47:14.158674-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 02:47:14.163653-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:14.165645-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:14.167651-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-08-26 02:47:14.172655-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:14.174530-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:14.176397-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:47:14.180676-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:24.183478-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:24.184201-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:47:24.187301-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:34.747708-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:34.749177-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:47:34.753645-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:44.756280-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:44.757859-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:47:44.762244-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:47:54.764703-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:47:54.766347-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:47:54.770881-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:48:04.773419-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:48:04.774935-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:48:04.779483-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:48:14.781991-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:48:14.783670-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:48:14.788146-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:48:24.789842-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:48:24.789848-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:48:24.792780-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 02:48:34.792795-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 02:48:34.792800-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 02:48:34.795383-0500 0x238      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
```

tommyvct, 2020-08-26:
feel free to ask anything

tommyvct, 2020-08-26:
FYI this is the refernce log when it works
```
2020-08-26 03:17:55.507801-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-08-26 03:17:55.515000-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-08-26 03:17:55.517133-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-08-26 03:17:55.521364-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-08-26 03:17:55.525928-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-08-26 03:17:55.530303-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-08-26 03:17:55.532312-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-26 03:17:55.536615-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 03:17:55.545546-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 03:17:55.547491-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 03:17:55.552536-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 03:17:55.554487-0500 0x1ba      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-08-26 03:17:55.556412-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 03:17:55.558277-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-08-26 03:17:55.562551-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 03:17:55.564442-0500 0x1ba      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 38
2020-08-26 03:17:55.564703-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 03:17:55.566616-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-08-26 03:17:55.571546-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 03:17:55.573331-0500 0x1ba      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 38
2020-08-26 03:17:55.573585-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 03:17:55.575371-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-08-26 03:17:55.579550-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-26 03:18:06.529880-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-26 03:18:06.531788-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-08-26 03:18:06.533679-0500 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

```

maxhartung, 2020-08-27:
I can confirm the problem. The problem for me is that my Apple TrackPad and Keyboard are laggy with this kext and itlwm combined. I didn't tried it without itlwm.

`2020-08-27 10:03:51.970657+0300 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-08-27 10:03:51.970983+0300 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-08-27 10:03:53.125262+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-08-27 10:03:53.125273+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-08-27 10:03:53.125282+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-08-27 10:03:53.125357+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-08-27 11:37:30.436073+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: getConfigurationDescriptor(0) failed
2020-08-27 11:37:30.436077+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: init usb configuration failed
2020-08-27 11:37:30.436080+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-08-27 11:37:30.436313+0300 0x4142     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-08-27 11:37:30.436366+0300 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-08-27 11:37:30.810809+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-08-27 11:37:30.810819+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-08-27 11:37:30.810829+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-08-27 11:37:30.810906+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-08-27 11:37:32.311119+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-08-27 11:37:32.311125+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-08-27 11:37:32.311145+0300 0x41b7     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-08-27 11:37:32.311220+0300 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-08-27 11:37:32.654912+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-08-27 11:37:32.654924+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-08-27 11:37:32.654935+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-08-27 11:37:32.655045+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-08-27 11:37:34.156422+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-08-27 11:37:34.156902+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-08-27 11:37:34.156932+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-08-27 11:37:34.157154+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-08-27 11:37:34.157356+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-08-27 11:37:34.157553+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-08-27 11:37:34.157556+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-08-27 11:37:34.159382+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-08-27 11:37:34.170898+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-27 11:37:34.170902+0300 0x4271     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
[... cut ...]
2020-08-27 12:30:54.832031+0300 0x14b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-08-27 12:30:54.832047+0300 0x53b5     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-08-27 12:30:54.832061+0300 0x53b5     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-08-27 12:30:54.832065+0300 0x53b5     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...`

williambj1, 2020-08-27:
Sorry for the late reply, I was super busy these days :p

@tommyvct Hmmm, this is odd, does this issue occur when you manually load itlwm after booting into the system with the Bluetooth driver injected by the bootloader? 

BTW, which bootloader are you using?

---

@maxhartung I think your issue is not the same as @tommyvct's. It's basically a physical coex issue and we cannot implement a workaround ATM unfortunately.

tommyvct, 2020-08-27:
It seems that sideload with SIP disabled works fine. But with caveats.

I use OpenCore 0.59 as by bootloader.
It used to work well before I use itlwm, but after I used itlwm it will fail uploading the firmware (especially if the computer is recovered from a kernel panic or forced shutdown). I disabled itlwm and the problem didn't really went away. I still need to reboot from Windows to make it work. I also have openSUSE and Ubuntu on my computer, will test if restarted from Linux improves anything.

I also tried @usr-sse2's Black80211-Catalina, it seems that the version of itlwm that come with it will made things worse, including kernel panics and IO hang.

I end up loading itlwm from the bootloader along with this firmware uploader.

tommyvct, 2020-08-27:
https://github.com/OpenIntelWireless/itlwm/issues/272

I also filed another issue in itlwm repo, hope it helps.

tommyvct, 2020-08-27:
I tried to trigger a kernel panic manually, it didn't kill the firmware uploader. Weird.

It seems that only panic triggered by itlwm will ruin it.

I also noticed that in the log I gave earlier, the uploader struggled on a particular action, probably caused by inconsistencies from the panic, failing the pre&post conditions of that uploading step.

mikasaloli, 2020-09-02:
same boat, i cant use with itlwm also

maxhartung, 2020-09-02:
> same boat, i cant use with itlwm also

I did some more tests. The Bluetooth is no longer working after using these two kexts. It gets back only when I boot into Windows and I reset the bluetooth module from Mac

yamileon, 2020-09-09:
getting the same issue on Mojave 10.14.6 

paste bin of the logs
https://pastebin.pl/view/6d7e1e23

tommyvct, 2020-09-14:
I recently updated opencore to 0.6.1.

I injected these kexts with opencore with this order:
[image]
Both the bluetooth and `itlwm` worked without any problem.

williambj1, 2020-09-14:
Glad that it works.

closed 2020-09-14 by williambj1