OpenIntelWireless/IntelBluetoothFirmware #180: AirPods Pro 在接QQ语音时没有声音,麦克风也没声音,系统面板显示所选设备没有输出控制

qhhuang, 2020-09-04:
**Description**
> A clear and concise description of what the bug is

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: ax201
- Product ID: 
- macOS Version: 10.15.6 (19G2021)

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple蓝牙软件版本：	7.0.6f7
  硬件、功能和设置：
  名称：	qhhuang’s iMac
  地址：	54-8D-5A-20-01-E7
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	10.256 (10.256)
  蓝牙电源：	打开
  可被发现：	关闭
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0026
  蓝牙核心规范：	Unknown (0xA)
  HCI修正版：	0x100
  LMP版本：	Unknown (0xA)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Desktop
  复合类设备：	0x380104
  设备类（主要）：	0x01
  设备类（次要）：	0x01
  服务类：	0x1C0
  自动寻找键盘：	打开
  设备（已配对、已配置等）：
  lazhazha：
  地址：	F0-5C-D5-15-AB-2A
  主要类型：	Audio
  次要类型：	Headphones
  服务：	Audio Sink, AVRCP Controller, AAP Server, Handsfree, AVRCP Target
  已配对：	是
  已配置：	是
  已连接：	是
  生产企业：	Apple (0x9, 0x7D31)
  蓝牙核心规范：	5.0
  固件版本：	2​D​27
  供应商ID：	0x004C
  产品ID：	0x200E
  设备类：	0x04 0x06 0x240418
  RSSI：	0
  角色：	主设备
  连接模式：	活跃模式
  时间间隔：	0 ms
  支持EDR：	是
  支持eSCO：	是
  支持SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM通道：	3
  需要认证：	否

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
2020-09-04 10:19:41.714414+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-09-04 10:19:41.714672+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-09-04 10:19:41.714681+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-09-04 10:19:41.714882+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-09-04 10:19:41.715103+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-09-04 10:19:41.715322+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-09-04 10:19:41.715324+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-09-04 10:19:41.715458+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-04 10:19:41.716098+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.4 build 0 week 30 2018
2020-09-04 10:19:41.716101+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-19-0-4.sfi
2020-09-04 10:19:41.716105+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-04 10:19:41.716107+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-09-04 10:19:41.716229+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-04 10:19:41.717100+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 2
2020-09-04 10:19:41.717102+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-09-04 10:19:41.717103+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-09-04 10:19:41.717104+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-09-04 10:19:41.717105+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-09-04 10:19:41.717106+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-09-04 10:19:41.717218+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-19-0-4.sfi
2020-09-04 10:19:41.717222+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-04 10:19:41.717224+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-09-04 10:19:41.717542+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-09-04 10:19:41.718087+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-09-04 10:19:41.718773+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-09-04 10:19:43.072036+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x24800
2020-09-04 10:19:43.072196+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-09-04 10:19:43.072197+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-04 10:19:43.072198+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-09-04 10:19:43.072332+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-09-04 10:19:43.073085+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
2020-09-04 10:19:43.073086+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-09-04 10:19:43.073092+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-04 10:19:43.086089+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Device reboot done
2020-09-04 10:19:43.086090+0800 0x15a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-09-04 10:19:43.086094+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-04 10:19:43.086096+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-09-04 10:19:43.086235+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-04 10:19:43.087088+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-04 10:19:43.087092+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-09-04 10:19:43.087093+0800 0x18e      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.
图1是 在语音时  图2是正常时
在语音时直接网易云音乐也没声音了
[image]
[image]

Menchen, 2020-09-08:
目前麦克风没办法，出现没声音的时候切换到内置扬声器然后换到蓝牙。

frozenzero123, 2020-12-19:
我也遇到了同样的问题

aluveitie, 2021-06-08:
The Intel Bluetooth Firmware is not compatible with Apples Bluetooth drivers in that regard, Microphone won't work.

closed 2021-06-22 by zxystd