OpenIntelWireless/IntelBluetoothFirmware #183: Issue: Can not load kext

HsunGong, 2020-09-08:
**Description**
> A clear and concise description of what the bug is

**Environment**

- Kext Version: 1.12
- WiFi Card Model: AC8265 
- Product ID: Thinkpad x1c6th
- macOS Version: macaos 10.15.6

**Info in System Report - Bluetooth**

No Log returned

**Logs**

No log returned.

**Screenshots**
> If applicable, add screenshots to help explain your problem.

[image]
[image]
[image]

meteoro, 2020-09-10:
I have same problem, start mac os with opencore after windows the driver is load and the log is possible:

`Firmware: Driver Stop()
2020-09-10 00:16:17.815393-0300 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-09-10 00:16:18.023621-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-09-10 00:16:18.023631-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:16:18.023639-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:16:18.023713-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-09-10 00:16:19.524542-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-09-10 00:16:19.524546-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-09-10 00:16:19.524554-0300 0x2889     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-09-10 00:16:19.524604-0300 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-09-10 00:16:19.732140-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-09-10 00:16:19.732149-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:16:19.737400-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:16:19.737468-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-09-10 00:16:21.237544-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-09-10 00:16:21.237548-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-09-10 00:16:21.237556-0300 0x289f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-09-10 00:16:21.237603-0300 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-09-10 00:16:21.444313-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-09-10 00:16:21.444321-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:16:21.449473-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:16:21.449541-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-09-10 00:16:22.951060-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: start fail, can not open device
2020-09-10 00:16:22.951065-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-09-10 00:16:22.951081-0300 0x28b8     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-09-10 00:16:22.951142-0300 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-09-10 00:16:23.231536-0300 0x28c9     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-09-10 00:16:23.231543-0300 0x28c9     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:16:23.234543-0300 0x28c9     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:16:23.234667-0300 0x28ba     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-09-10 00:16:23.646537-0300 0x28cf     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-09-10 00:16:23.646546-0300 0x28cf     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:16:23.651506-0300 0x28cf     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:16:23.651568-0300 0x28cf     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
[... cut ...]
2020-09-10 00:17:00.753227-0300 0x2b67     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-09-10 00:17:00.758643-0300 0x2b67     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-09-10 00:17:00.758705-0300 0x2b67     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()`

Drowningfish223, 2020-10-01:
I have the same problem. However, it's not a top priority issue. I don't mind having to boot windows occasionally if I need to get bluetooth that way. Thanks for your work, devs!

ShreyansJa1n, 2020-10-05:
Have the same issue!

signorettif, 2020-10-08:
Same issue for me on NUC 8 i5

williambj1, 2020-10-08:
Obvious user mistake.

closed 2020-10-08 by williambj1