OpenIntelWireless/IntelBluetoothFirmware #186: Slow kext loading

nemphys, 2020-09-16:
The kext works fine on my NUC8i3BEH, but it takes more than 10 seconds to load during boot, which inevitably slows boot up time.
I am wondering if something can be done in order to speed things up...

MichaelDevC, 2020-10-14:
I also have the same problem... slow loading of the kext and also of all the boot of my hackintosh

Please solve it

profzei, 2020-10-15:
In your `config.plist` set to `false` the loading of `IntelBluetoothInjector.kext`

williambj1, 2020-10-16:
The USB part of the card really takes some time to respond, uploading firmware through USB 2.0 is quite slow as well. Really can't improve.

MichaelDevC, 2020-10-16:
> In your `config.plist` set to `false` the loading of `IntelBluetoothInjector.kext` 

And after how I can use the ON/OFF native switch of MacOS?

nemphys, 2020-12-25:
I am wondering, if it is really a hardware limitation, how come Windows/Linux manage to boot in 5-10 seconds while performing the firmware upload? It truly makes to sense.

simon-sino, 2021-04-30:
my nuc5 takes more than 2mins to pass this cycle:

intelfirmware:interrupt continue
intelfirmware:hci_op_intel_version
intelfirmware:interrupt 

set false to the loading of IntelBluetoothInjector.kext  
can boot fast

closed 2021-06-22 by zxystd