OpenIntelWireless/IntelBluetoothFirmware #187: 使用1.1.2蓝牙驱动和v1.1.0stable Wi-Fi驱动连接蓝牙耳机（先锋）播放音乐，网络异常的缓慢甚至断网，耳机声音断断续续

sumforest, 2020-09-18:
2020-09-18 21:49:53.049477+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-09-18 21:49:53.049635+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-09-18 21:49:53.049637+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-18 21:49:53.049639+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-09-18 21:49:53.049790+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-09-18 21:49:53.050741+0800 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
2020-09-18 21:49:53.050743+0800 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-09-18 21:49:53.050754+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-18 21:49:53.064755+0800 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Device reboot done
2020-09-18 21:49:53.064759+0800 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-09-18 21:49:53.064764+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-18 21:49:53.064765+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-09-18 21:49:53.065002+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-09-18 21:49:53.065740+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-18 21:49:53.065745+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-09-18 21:49:53.065747+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-09-18 22:41:24.444432+0800 0x66ff     Default     0x0                  2142   0    网易有道词典: selection: og show --last boot | grep IntelFirmware
2020-09-18 21:49:53.049477+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-09-18 21:49:53.049635+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-09-18 21:49:53.049637+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-09-18 21:49:53.049639+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-09-18 21:49:53.049790+0800 0x17d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-09-18 21:49:53.050741+0800 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
- 蓝牙设备：0x8087, 0x0AAA
- Wi-Fi型号：英特尔 Wireless-AC 9560

stevezhengshiqi, 2020-09-20:
请使用5G网络来规避2.4G信道上蓝牙和wifi的干扰。

closed 2020-09-20 by williambj1

sumforest, 2020-09-20:
感谢，回答