OpenIntelWireless/IntelBluetoothFirmware #192: Bluetooth doesn't show (OC 0.6.1, AC 3168)

hlcfan, 2020-10-04:
**Description**
> Bluetooth not show, WiFi works along with using HeliPort. Both `itlwm` and `IntelBluetoothFirmware` are placed in CO Kexts. However from `Hackintool`, `IntelBluetoothFirmware` isn't installed.

[image]
 
BTW, do I need a USB dongle to make the bluetooth to work or it's natively supported by `IntelBluetoothFirmware`?

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: Intel® Dual Band Wireless-AC 3168 module 
- Product ID: 
- macOS Version: Catalina 10.15.7
- Motherboard Chipset: AMD B450

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between

```
No information found.
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
EMPTY
```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```
NO PANIC
```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

[image]

[image]

williambj1, 2020-10-04:
Map your USB

closed 2020-10-04 by williambj1

Kaijun, 2020-10-05:
> Map your USB

sorry, could you provide more details about the usb mapping for bluetooth?

my device-id is 0x8087, 0x0a2a, which is encoutering the same problem with @hlcfan 。kext not loaded and no log output totally.

i also checked my XHCI devices in ioreg, i didn't find anything about bluetooth in usb devices.