OpenIntelWireless/IntelBluetoothFirmware #205: 找不到蓝牙，驱动已加载，见log

Babyvox550, 2020-10-20:
**Description**
<找不到蓝牙，驱动已加载>

**Environment**

- Kext Version: 1.1.2
- WiFi Card Model: INTEL AC9560
- Product ID: 
- macOS Version: 10.15.7

**Info in System Report - Bluetooth**
2020-10-20 22:03:24.540581+0800 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-10-20 22:03:24.541245+0800 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-10-20 22:03:24.545509+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-10-20 22:03:24.545520+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-10-20 22:03:24.545528+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-10-20 22:03:24.545598+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-10-20 22:03:26.049539+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-10-20 22:03:26.050661+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-10-20 22:03:26.050825+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-10-20 22:03:26.051762+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-10-20 22:03:26.052388+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-10-20 22:03:26.052908+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-10-20 22:03:26.052916+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-10-20 22:03:26.074519+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-20 22:03:26.075283+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.1 build 0 week 30 2018
2020-10-20 22:03:26.075289+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-19-0-1.sfi
2020-10-20 22:03:26.075300+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-20 22:03:26.075303+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-10-20 22:03:26.075465+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-20 22:03:26.076328+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 2
2020-10-20 22:03:26.076332+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-10-20 22:03:26.076334+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-10-20 22:03:26.076336+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-10-20 22:03:26.076343+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-10-20 22:03:26.076346+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-10-20 22:03:26.076533+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-19-0-1.sfi
2020-10-20 22:03:26.076541+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-20 22:03:26.076543+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-10-20 22:03:26.076946+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-10-20 22:03:26.077612+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-10-20 22:03:26.078435+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-10-20 22:03:30.588108+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-10-20 22:03:30.593413+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-10-20 22:03:30.593426+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-20 22:03:30.593434+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-10-20 22:03:30.593807+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-10-20 22:03:30.594638+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Notify: Firmware download done
2020-10-20 22:03:30.594654+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-10-20 22:03:30.594729+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-20 22:03:30.884280+0800 0x17b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: onRead not responding
2020-10-20 22:03:30.884382+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-20 22:03:30.884405+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-10-20 22:03:30.885054+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Setting Intel event mask failed (-536870184) device is not ready
2020-10-20 22:03:30.885081+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-10-20 22:03:30.885089+0800 0x5846     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-10-20 22:03:30.888035+0800 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-10-20 22:03:30.889139+0800 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-10-20 22:03:30.893421+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-10-20 22:03:30.893431+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-10-20 22:03:30.893441+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-10-20 22:03:30.893535+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-10-20 22:03:32.396358+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
[... cut ...]
2020-10-20 22:32:50.517051+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-10-20 22:32:50.517059+0800 0x587f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-10-20 22:32:50.517631+0800 0x59c3     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: onRead unhandle status (-536870165) operation was aborted)

**Screenshots**
[image]
[image]
[image]
[image]

本人小白，请提供解决步骤，谢谢！

williambj1, 2020-10-20:
蓝牙固件上传失败，如果直接从 Windows 或 Linux 热启动进 macOS 蓝牙是否可用？

Babyvox550, 2020-10-22:
https://github.com/Babyvox550/xiaomi15PRO-11-OC-EFI/releases/tag/V0.0.2 

我稀里糊涂的就完美使用了，这个是目前的EFI，适用BigSur B9。蓝牙已经能连接音响，wifi直连。HiDPi也没有1/4屏了。
只有内置音响问题了。
另外，隔空投送还是不能用。

zxystd, 2020-10-30:
问题解决就行，另外没有人说过现在隔空投送是可以用的。感谢反馈。

closed 2020-10-30 by zxystd