OpenIntelWireless/IntelBluetoothFirmware #211: 蓝牙工作不正常，可以搜索、发送文件，无法接收，无法连接鼠标

EvilZabanya, 2020-10-30:
系统信息
`  Apple蓝牙软件版本：	7.0.6f7
  硬件、功能和设置：
  名称：	Zero-MAC的MacBook Pro
  地址：	18-1D-EA-CD-08-83
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	8.4096 (8.4096)
  蓝牙电源：	打开
  可被发现：	打开
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0A2A
  蓝牙核心规范：	4.2 (0x8)
  HCI修正版：	0x1000
  LMP版本：	4.2 (0x8)
  LMP子版本：	0x1000
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Portable
  复合类设备：	0x38010C
  设备类（主要）：	0x01
  设备类（次要）：	0x03
  服务类：	0x1C0
  自动寻找键盘：	打开`

log
`2020-10-30 13:59:21.440551+0800 0x6f       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-10-30 13:59:21.440897+0800 0x6f       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()
2020-10-30 13:59:21.445833+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-10-30 13:59:21.445849+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-10-30 13:59:21.445861+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2A
2020-10-30 13:59:21.445983+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-10-30 13:59:22.948706+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-10-30 13:59:22.949652+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found interface!!!
2020-10-30 13:59:22.949718+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Interrupt endpoint!
2020-10-30 13:59:22.950039+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Bulk out endpoint!
2020-10-30 13:59:22.950315+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found Bulk in endpoint!
2020-10-30 13:59:22.950552+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-10-30 13:59:22.950555+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_RESET
2020-10-30 13:59:22.951309+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-10-30 13:59:22.964471+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:22.964483+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-10-30 13:59:22.964847+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-10-30 13:59:22.965017+0800 0x178      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found device firmware ibt-hw-37.8.10-fw-1.10.3.11.e.bseq
2020-10-30 13:59:22.965140+0800 0x178      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: request firmware success
2020-10-30 13:59:22.965162+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:22.965169+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_ENTER_MFG
2020-10-30 13:59:22.965502+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-10-30 13:59:22.966182+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:23.270942+0800 0x178      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-10-30 13:59:23.272233+0800 0x178      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-10-30 13:59:23.275254+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:23.275266+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EXIT_MFG
2020-10-30 13:59:23.275582+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-10-30 13:59:23.289352+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:23.289364+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-10-30 13:59:23.289736+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-10-30 13:59:23.290053+0800 0x178      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-10-30 13:59:23.290081+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-10-30 13:59:23.290105+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-10-30 13:59:23.290112+0800 0x180a     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
`

williambj1, 2020-10-30:
Issue out of scope, read our docs for more info.

closed 2020-10-30 by williambj1

zxystd, 2020-10-30:
去设置里面打开共享才能接收文件，就算是白果也是要这样的。

EvilZabanya, 2020-10-30:
> 去设置里面打开共享才能接收文件，就算是白果也是要这样的。

妈耶。开启共享以后能蓝牙传输文件了，但是蓝牙鼠标还是不行
驱动部分显示4096，不知道是哪里不对