OpenIntelWireless/IntelBluetoothFirmware #212: I don't get my bluetooth to work. Some times it works after sleep.

orveomindgrind, 2020-10-31:
**Description**
I don't get my bluetooth to work. Some times it works after sleep.

**Environment**

- Kext Version: Latest 
- WiFi Card Model:

 Ethernet Chipset: Realtek RTL8125B 2.5Gb Ethernet
Wireless Chipset: Intel® Wi-Fi 6 AX200
Bluetooth: Bluetooth v5.1*

- Product ID: Catalina
- macOS Version: 10.15.7

**Logs**

Last login: Sat Oct 31 15:21:26 on ttys000
orveomindgrind@Desktop-oSXs-iMac-Pro ~ % log show --last boot | grep IntelFirmware
2020-10-31 15:21:06.043020+0100 0x28f      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI Timeout, retry
2020-10-31 15:21:06.045483+0100 0x28f      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-31 15:21:06.047821+0100 0x28f      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET_BL
2020-10-31 15:21:06.270055+0100 0x28f      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-10-31 15:21:06.272491+0100 0x28f      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-10-31 15:21:08.266036+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-10-31 15:21:08.273457+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-10-31 15:21:08.274911+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-10-31 15:21:08.279889+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-10-31 15:21:08.281992+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-10-31 15:21:08.284484+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-10-31 15:21:09.795204+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-10-31 15:21:09.803226+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-10-31 15:21:09.806220+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-10-31 15:21:09.810935+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-10-31 15:21:09.816086+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-10-31 15:21:09.821485+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-10-31 15:21:09.823629+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-10-31 15:21:09.828202+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-31 15:21:09.830108+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.0 build 109 week 20 2020
2020-10-31 15:21:09.830373+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-20-0-0.sfi
2020-10-31 15:21:09.832402+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-10-31 15:21:09.834381+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-31 15:21:09.836306+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-10-31 15:21:09.841211+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-31 15:21:19.845561+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI Timeout, retry
2020-10-31 15:21:19.848009+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-31 15:21:19.850316+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET_BL
2020-10-31 15:21:20.073760+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-10-31 15:21:20.076090+0100 0x7d6      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-10-31 15:21:20.083041+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-10-31 15:21:20.087777+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-10-31 15:21:21.565322+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-10-31 15:21:22.046591+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-10-31 15:21:22.078783+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-10-31 15:21:22.171273+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-10-31 15:21:23.724782+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-10-31 15:21:24.424287+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2020-10-31 15:21:24.459647+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2020-10-31 15:21:24.506340+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2020-10-31 15:21:24.551025+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2020-10-31 15:21:24.594712+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-10-31 15:21:24.627970+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-10-31 15:21:24.669290+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-10-31 15:21:24.670280+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.0 build 109 week 20 2020
2020-10-31 15:21:24.761729+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-20-0-0.sfi
2020-10-31 15:21:24.819888+0100 0x233      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-10-31 15:21:24.865589+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-10-31 15:21:24.901929+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-10-31 15:21:24.946290+0100 0xa12      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
[... cut ...]
2020-10-31 15:23:28.215311+0100 0x242b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-10-31 15:23:28.244442+0100 0x242b     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2020-10-31 15:23:28.273632+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-10-31 15:23:28.304714+0100 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()

williambj1, 2020-11-02:
Nothing wrong according to the logs, try to inject the kext as late as possible.

MammaScan, 2020-11-10:
Same problem, How do I delay the injection of kext other than edit the config.plist so that the kext will be the last in the kext list?
 %  log show --last boot | grep IntelFirmware
2020-11-10 08:42:15.169140+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:42:15.183870+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:42:15.184257+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:42:25.185553+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:42:25.204029+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:42:25.217949+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:42:35.226586+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:42:35.239096+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:42:35.245981+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:42:45.253564+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:42:45.263240+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:42:45.270737+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:42:55.281064+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:42:55.289506+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:42:55.296104+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:05.301692+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:05.310812+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:05.317694+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:15.324001+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:15.324008+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:15.324285+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:25.324352+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:25.324354+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:25.324504+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:35.325002+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:35.325009+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:35.325210+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:45.325242+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:45.325266+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:45.325448+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:43:55.325560+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 08:43:55.325562+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-11-10 08:43:55.325699+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-11-10 08:44:05.326244+0100 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
[... cut ...]
2020-11-10 11:01:44.433727+0100 0x16d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-11-10 11:01:44.433732+0100 0x1450d    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-11-10 11:01:44.433736+0100 0x1450d    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-11-10 11:01:44.433738+0100 0x1450d    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

AndryX19, 2020-11-10:
I have the same problem, no more bluetooth after latest update of itlwm kext for wifi intel

lgariv, 2020-11-15:
hmm... was solved for me in a previous commit and now on the latest one it is happening again as well. 9462, bluetooth product id `0x0AAA`

closed 2021-06-22 by zxystd