OpenIntelWireless/IntelBluetoothFirmware #213: 蓝牙鼠标连接成功但无法使用

SnowsonZ, 2020-10-31:
**Description**

非常感谢该项目，帮助我解决了蓝牙和wifi的驱动问题。现遇到1个关于蓝牙的问题，我的蓝牙鼠标可以正常连接，但是连接之后却无法使用。
**Environment**
 
- Kext Version: 1.1.2
- WiFi Card Model: 
- Product ID: 0x8087 | 0x0A2B
- macOS Version:   Catalina 10.15.2 (19C57)

**Info in System Report - Bluetooth**

```

```

**Logs**

```log
2020-11-01 04:18:02.096293+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: boot_param=0x40800
2020-11-01 04:18:02.096456+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware done
2020-11-01 04:18:02.096458+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-11-01 04:18:02.096459+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_RESET
2020-11-01 04:18:02.096593+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Intel reset succeed
2020-11-01 04:18:02.096759+0800 0x14d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Notify: Firmware download done
2020-11-01 04:18:02.096761+0800 0x14d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-11-01 04:18:02.096770+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-11-01 04:18:02.108783+0800 0x14d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Notify: Device reboot done
2020-11-01 04:18:02.108786+0800 0x14d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-11-01 04:18:02.108793+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-11-01 04:18:02.108795+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-11-01 04:18:02.108969+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-11-01 04:18:02.117793+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-11-01 04:18:02.117802+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-11-01 04:18:02.117804+0800 0x192      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
```

**Kernel Panic Logs**

```

```

**Screenshots**

[image]

SnowsonZ, 2020-11-02:
今天发现airpods 2代语音不可用

williambj1, 2020-11-02:
蓝牙耳机麦克风问题已知。两个问题都与固件上传无关，超出了本驱动的管理范围，以后可能会通过其它方式提供修补。详情请阅读官方文档。

closed 2020-11-02 by williambj1