OpenIntelWireless/IntelBluetoothFirmware #217: Intel BTLE 9460/9560 Module loses connectivity with mouse/keyboard with no reason

mackonsti, 2020-11-11:
**Description**
I am running latest Catalina on my **Intel NUC8i7BEH** mini PC that has integrated the Intel Bluetooth 9460/9560 Module (BTLE v5.1) device [8087:0aaa] and this BTLE has PCI ID is https://pci-ids.ucw.cz/read/PC/8087/0aaa
When attaching a BT keyboard or mouse, suddenly it goes disconnected without apparent reason. I cannot find a pattern in the disconnections. The mouse, especially, doesn't cooperate at all. Both work 100% without issues on my work's MacBookPro15,4.

**Environment**
- Kext Version: Stable 1.1.0 from your Releases
- WiFi Card Model: The NUC has the combo WLAN/BTLE card Intel Dual-Band Wireless-AC 9560
- Product ID: [8087:0aaa] per Hackintool; the WLAN card is [8086:9df0]
- macOS Version: Catalina 10.15.7

**Info in System Report - Bluetooth**
```
Apple Bluetooth Software Version:	7.0.6f7
  Hardware, Features and Settings:
  Name:	Gray Mini
  Address:	38-BA-F8-93-99-7C
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AAA
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  Satechi Slim Keyboard:
  Address:	21-18-08-08-1E-0E
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Broadcom Bluetooth Wireless Keyboard SDP Server, Broadcom Bluetooth Wireless  Keyboard                        , Broadcom Bluetooth Wireless Keyboard PnP Server
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x2211)
  Bluetooth Core Spec:	3,0
  Firmware Version:	0x011B
  Vendor ID:	0x2222
  Product ID:	0x0035
  Class of Device:	0x05 0x10 0x0540
  AFH:	On
  AFH Map:	CEFF4000200C0730
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	No
  EDR Supported:	No
  eSCO Supported:	No
  SSP Supported:	No
  Konstantino's AirPods:
  Address:	24-F6-77-F0-16-7D
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	
[... 95 lines cut ...]
2020-11-11 13:24:10.500201+0100 0x156      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Notify: Firmware download done
2020-11-11 13:24:10.500206+0100 0x156      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-11-11 13:24:10.500243+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-11-11 13:24:10.513206+0100 0x156      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Notify: Device reboot done
2020-11-11 13:24:10.513209+0100 0x156      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-11-11 13:24:10.513215+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-11-11 13:24:10.513217+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-11-11 13:24:10.513422+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-11-11 13:24:10.514302+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-11-11 13:24:10.514333+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-11-11 13:24:10.514335+0100 0x574      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Clean up...
```

**Kernel Panic Logs**
No kernel panic.

**Screenshots**
No useful screenshots, really.

**Additional information**
I have the impression that this firmware uploaded for this BTLE device doesn't like my Bluetooth devices.
My mouse is a **Logitech MX Anywhere 2** model and my wireless keyboard is a **Satechi Aluminum Wireless Keyboard** model ST-TCAWKS / 879961007249 according to their website.
Happy to provide supplemental logs if needed. Thank you.

[image]

lgariv, 2020-11-14:
AC-9462, bought a Logitech Triathlon M720 mouse recently, and can't even pair it to begin with, so no logs available.

It says "Connected" in the bluetooth menu, with a spinning wheel, and then it just disappears after about 7 seconds. The light on the mouse is blinking all of the time, meaning it never actually pair.

I've done some digging around, and this problem is most probably related to BLE - I just don't no where to even start to look into in order to fix that (or at least try).

williambj1, 2020-11-15:
Hi @mackonsti @lgariv!

I appreciate your detailed but report, but unfortunately, this issue is currently out of our control. IntelBluetoothFirmware is just a firmware uploader. Due to our limited time, we are focusing on itlwm and not planning on solving this. We are looking forward to external contributions. Please read our docs for more info. Sorry for the inconvenience.

closed 2020-11-15 by williambj1

lgariv, 2020-11-15:
> Hi @mackonsti @lgariv!
>
>
>
> I appreciate your detailed but report, but unfortunately, this issue is currently out of our control. IntelBluetoothFirmware is just a firmware uploader. Due to our limited [...]

If you are not going to fix it yourself but are looking for a contribution, why closing this issue? Should be pinned in my opinion so other developers capable of fixing the issue my see it.

zxystd, 2020-11-15:
@lgariv Nothing wrong with this driver, this **firmware uploader** is a functionality driver. The problems @mackonsti described is related to Apple's driver, need a patch driver or write a entire HID driver if you are willing to do, they are different drivers. If I have more time and decide to fix these compatibility problems, I will create another driver repo, nothing can be done base on this driver. It works fine.

williambj1, 2020-11-15:
> If you are not going to fix it yourself but are looking for a contribution, why closing this issue? Should be pinned in my opinion so other developers capable of fixing the issue might see it.

@lgariv I closed this issue mainly because it a duplication of #51, which is already pinned.

williambj1, 2020-11-15:
Duplicate of #51